raspiblitz/raspiblitz #2369: Not detecting umbrel when attempting to migrate over?

DocCodez, 2021-06-23:
I was on umbrel 0.3.12 and was looking to migrate to raspiblitz and was following the instruction but it is not detecting a migration from umbrel.

rootzoll, 2021-06-23:
Thanks for the report - will retest for next release. 
Do you have any special configuration (other hadware then recommended or had some additional software)?

For now you should be able to just reinsert the umbrel sd card again and it should continue running.

rootzoll, 2021-06-24:
I am preparing a Umbrel 0.3.13 for testing ... it should be the same as the 0.3.12 - just with updated containers.

AaronDewes, 2021-06-26:
We didn't really change anything related to the mount/file system logic recently in Umbrel, so I'm not sure what went wrong in your case @DocCodez, we don't try to make anything in Umbrel to prevent you from migrating to a different solution.

rafg100, 2021-06-26:
I performed a new install of raspiblitz today and migration from umbrel v0.3.13 was successful.

nyxnor, 2021-06-26:
@DocCodez Reflash the sd card and try again please. About not detecting, are you share it got to that point that mentions on how it is gonna sync the blockchain or reuse an existent one? Wait for this menu instructions.

rootzoll, 2021-09-20:
Tested the Umbrel migration for v1.7.1RC2 ... looked good. Putting this on final testing.

rootzoll, 2021-09-21:
OK tested again and working. Closing issue.

closed 2021-09-21 by rootzoll