raspiblitz/raspiblitz #2376: build_sdcard.sh get stuck at grpcio install

atef-zayati, 2021-06-26:
Hello everyone,

I'm installing raspiblitz via `build_sdcard.sh`. The script get stuck at this line : 
`sudo -H python3 -m pip install grpcio==1.36.1`

And here is the output : 
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting grpcio==1.36.1
  Downloading https://files.pythonhosted.org/packages/46/49/aa254d395a594245cfa6c5c09eb5cf278edb9847babdb88efebba1ca86e4/grpcio-1.36.1.tar.gz (21.5MB)
    100% |████████████████████████████████| 21.5MB 16kB/s 
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio==1.36.1) (1.12.0)
Building wheels for collected packages: grpcio
  Running setup.py bdist_wheel for grpcio ... /`

Any ideas? How to proceed? Thank you.

nyxnor, 2021-06-26:
```
uname -a
df -Th
```

atef-zayati, 2021-06-26:
```
$ uname -a
Linux raspberrypi 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux
```

```
$ df -Th
Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       29G  3.1G   25G  11% /
devtmpfs       devtmpfs  326M     0  326M   0% /dev
tmpfs          tmpfs     456M     0  456M   0% /dev/shm
tmpfs          tmpfs     456M   18M  438M   4% /run
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
tmpfs          tmpfs     456M     0  456M   0% /sys/fs/cgroup
/dev/mmcblk0p1 vfat      253M   30M  223M  12% /boot
tmpfs          tmpfs      92M  4.0K   92M   1% /run/user/1000
tmpfs          tmpfs      32M     0   32M   0% /var/cache/raspiblitz
```

nyxnor, 2021-06-26:
Ok, nothing weird, just to check.
Try reflashing the raspios_arm64 image and running the scripts again.

atef-zayati, 2021-06-26:
I'm not sure you have read the update. I canceled grpcio operation and the script resumed. I think I will reinstall a newer version of grpcio.

nyxnor, 2021-06-26:
Be patientm, you are using RPI3, better to sell it and get a PI4.
I just used a script that needed to install grpcio and took a long time with PI4.

atef-zayati, 2021-06-29:
I solved this issue: 
```
pip install --upgrade pip
pip install grpcio==1.38.1
```
It took only few seconds to complete.

VIRTUTEDUCE, 2021-07-08:
> This solved the issue:
> 
> ```
> pip install --upgrade pip
> pip install grpcio==1.38.1
> ```
> 
> It took only few seconds to complete.

Hi, i'm having the exact same issue here. I tried your solution and both pip & grpcio successfully installed. 

However when I build again the sdcard.sh, it tries to reinstall grpcio in 1.36.1 instead of 1.38.1 and gets stuck again at the same step. 

Could you please tell me how you did proceed ? 

Thanks for that.

atef-zayati, 2021-07-11:
> Hi, i'm having the exact same issue here. I tried your solution and both pip & grpcio successfully installed.
>
> However when I build again the sdcard.sh, it tries to reinstall grpcio in 1.36.1 [...]

You can replace grpcio version in build_sdcard.sh script and run it again.

rootzoll, 2021-08-24:
Test builds of sd card for v1.7.1rc1 look good now.

rootzoll, 2021-10-10:
all release candidates builded corerct with the script in dev branch ... closing issue for v1.7.1 release and merge into v1.7 branch on release.