raspiblitz/raspiblitz #2378: Change fingerprint hash from sha1 to sha256

nyxnor, 2021-06-26:
With sha256 the menu width I needed to change from 74 to 100.
I will need to be editing the bonus.services files anyway to my Tor PR, should I change this too?

[image]
[image]

openoms, 2021-06-26:
The question is what does the browser show so we can double check?

nyxnor, 2021-06-26:
The browser shows both

openoms, 2021-06-26:
Good, then the shorter is easier to verify.

nyxnor, 2021-06-26:
Yes, but it is broken https://crypto.stackexchange.com/questions/3690/why-is-sha-1-considered-broken

nyxnor, 2021-06-26:
[image]
from torproject.org

openoms, 2021-06-26:
How about braking the hash in two lines? Would that make it difficult to copy?

Also check what happens in the default size terminal windows. If 100 character width fits in there there is no problem. If it is wrapped then needs more lines (height) or might be cut off together with the text underneath.

nyxnor, 2021-06-26:
with 74 (default) It brakes to the second line, but now I cant copy and paste the certificate, as there is blank space between the first and second line.
100 width is the right width for sha256.

[image]

openoms, 2021-06-26:
The problem is when the terminal window is not fullscreen, but small default.

part of the text will be cut off then.

We usually limit the text (even in git commits) to 80 characters for readability in these situations.

The 74 character + window borders just fit in.

I think dialog is already a 'just about acceptable UX' so shouldn't really grow all info windows like that to 100.

Copying only part of the hash should do it for the ones following this practice, what do you think?

nyxnor, 2021-06-26:
Yes, copying part of that hash is a workaround, not perfect per se.

What could be done is echoing the sha256 fingerprint in the terminal, just like the tor address is when calling `bonus.service.sh menu`

nyxnor, 2021-06-26:
[image]

Now you can just scan the QR code to connect to the service, this will show my apache2 debian default page.

Unfortunately, Tor Browser mobile does not work with Client Auth yet, so I need to warn about this https://gitlab.torproject.org/tpo/applications/tor-browser/-/issues/31672. But you can still scan the QR code on mobile and open in Tor Browser android as normal.

nyxnor, 2021-07-22:
@rootzoll Even though you marked for v1.8, reading this I prefer to close, open at will.
Reason:
* sha256 is larger than 80 for the msgbox, openoms recommended using just up to 80.
* web ui solves this, the fingerprint can be presented there.

openoms, 2021-09-19:
> [image]
> 
> Now you can just scan the QR code to connect to the service, this will show my apache2 debian default page.
> The fingerprint was hardcoded, as I dont use tls for that domain, but the function is ready in the script.
> 
> Unfortunately, Tor Browser mobile does not work with Client Auth yet, so I need to warn about this https://gitlab.torproject.org/tpo/applications/tor-browser/-/issues/31672. But you can still scan the QR code on mobile and open in Tor Browser android as normal.

Don't get me wrong, I do prefer the info shown on thsi screenshot over the 80 character insecure fingerprint in whiptail. Re-opening to be implemented non-urgently.