raspiblitz/raspiblitz #2380: WARden Terminal

nyxnor, 2021-06-26:
There is the web WebUI version and the terminal version https://github.com/pxsocs/warden_terminal

Requests to price use tor.

Succesfull install:
[image]

There is an API key which expired doing my tests:
[image]

Due to having to deal with key expiration, I'm not sure if I should submit a PR for this.

Here it is the `bonus.warden_terminal.sh`

```bash
#!/bin/bash

# https://github.com/pxsocs/warden_terminal

# command info
if [ $# -eq 0 ] || [ "$1" = "-h" ] || [ "$1" = "-help" ]; then
 echo "config script to install or uninstall WARden Terminal"
 echo "bonus.warden_terminal.sh [on|off|menu]"
 exit 1
fi

source /mnt/hdd/raspiblitz.conf

# add default value to raspi config if needed
if ! grep -Eq "^warden_terminal=" /mnt/hdd/raspiblitz.conf; then
  echo "warden_terminal=off" >> /mnt/hdd/raspiblitz.conf
fi

# show info menu
if [ "$1" = "menu" ]; then
  dialog --title " Info WARden Terminal " --msgbox "
WARden Terminal is a command line tool.
Type: 'wt' in the command line to switch to the dedicated user.
Usage: https://github.com/pxsocs/warden_terminal/blob/master/README.md
" 10 75
  exit 0
fi

# install
if [ "$1" = "1" ] || [ "$1" = "on" ]; then

  echo "*** INSTALL WARDEN TERMINAL ***"

  # create wt user
  sudo adduser --disabled-password --gecos "" wt

  # download source code
  sudo -u wt git clone https://github.com/pxsocs/warden_terminal.git /home/wt/warden_terminal

  sudo -u wt pip3 install -r /home/wt/warden_terminal/requirements.txt
  sudo -u wt python3 /home/wt/warden_terminal/node_warden.py

  # setting value in raspi blitz config
  sudo sed -i "s/^warden_terminal=.*/warden_terminal=on/g" /mnt/hdd/raspiblitz.conf

  echo "# Usage: https://github.com/pxsocs/warden_terminal/blob/master/README.md"
  echo "# To start type: 'sudo su wt' in the command line."
  echo "# To exit the user - type 'exit' and press ENTER"

  exit 0

fi

# switch off
if [ "$1" = "0" ] || [ "$1" = "off" ]; then

  # setting value in raspi blitz config
  sudo sed -i "s/^warden_terminal=.*/warden_terminal=off/g" /mnt/hdd/raspiblitz.conf

  echo "*** REMOVING WARDEN TERMINAL ***"
  sudo userdel -rf wt
  echo "# OK, wt is removed."
  exit 0

fi
```

nyxnor, 2021-06-26:
PXSocs explained the key provided by default is limited, one should create a free account to request for more keys that will have 100k calls. 250k the key ends and need to create another account that just requires e-mail. https://min-api.cryptocompare.com/pricing

Their license is for personal use, so don't know if you can sell this https://www.cryptocompare.com/free-api-use/

[image]

pxsocs, 2021-06-26:
Thanks for this @nyxnor 
As a separate note, working on an upgrade that will include Raspberry pi specific info like: Temperature, voltage, cpu usage and others.

nyxnor, 2021-07-26:
https://twitter.com/alphaazeta/status/1419667516922765316/photo/1

pxsocs, 2021-07-31:
Latest Version has been tested on RaspiBlitz. Features:
. Auto-detects if running at a RaspiBlitz node
. Outputs to `/dev/tty1` or a different output by editing `config.ini`
. Automatically detects Bitcoin RPC setting

Would be great to get additional testers to run and provide feedback. 

https://github.com/pxsocs/warden_terminal

nyxnor, 2021-08-01:
@pxsocs Working. Liking the auto scrolling and slides type.

The script detects that I am using Raspiblitz, but this variable is not printed correct here: https://github.com/pxsocs/warden_terminal/blob/1ef6727b20885ce747e1521b13cd58ae40611d09/node_warden.py#L477

pxsocs, 2021-08-01:
Fixed for next version. Missed an `f` string :)

rootzoll, 2022-09-29:
OK. Sorry for having pushed this so long ahead - the RaspiBlitz WebUI was sucking up all my attention. So whats the status of WARDen RaspiBlitz integration? I dont see a PR for the Terminal Version and the WebUI PR #1678 & #1675 looks unmaintained and still based on RaspiBlitz v1.6 branch. Looking to release RaspiBlitz v1.8.1 end of October ... let me know if it still makes sense add this for the release.