raspiblitz/raspiblitz #2381: error on search address in mempool

GabrielRCL, 2021-06-26:
I can't search for any address in the mempool, it always gives this error:
Error: Connection to server lost, please retry

this error only happens with addresses. (blocks and transactions I can search normally)

[image]

openoms, 2021-06-26:
You need to have the Electrum Server: Electrs installed to search addresses. Give it a day to build the index and try again.

nyxnor, 2021-06-27:
```
Jun 27 00:46:03 raspberrypi electrs[1764]: 2021-06-27T00:46:03.217+01:00 - INFO - best=0000000000000000000526132e9400bcad0e912014c2ab98b8a4273ca6209fbe height=688930 @ 2021-06-26T23:45:35Z (1 left to index)
```

openoms, 2021-06-27:
Try a full restart first of all.

GabrielRCL, 2021-06-27:
I Reboot in my raspiblitz and now the error is this
[image]

openoms, 2021-06-27:
we use 1000 as a limit: https://github.com/rootzoll/raspiblitz/blob/a22589c86109d56ecc1e1aca7abb214c7e9189c7/home.admin/config.scripts/bonus.electrs.sh#L311

Can be edited in : `/home/electrs/.electrs/config.toml` , but first try some other address.

closed 2021-06-28 by GabrielRCL