raspiblitz/raspiblitz #2383: Tor docs

nyxnor, 2021-06-28:
Will need to enhance Tor docs, possibly a .md file just for it.

TPO issues:
* [You should make every Tor user be a relay](https://gitlab.torproject.org/tpo/web/support/-/issues/226) - Not possible at the moment and plenty of issues to be analyzed. This is 1 more reason why being a Tor relay can't be integrated with Blitz at the moment
* [Tor Browser for Android: Onion Service authentication](https://gitlab.torproject.org/tpo/applications/tor-browser/-/issues/31672) - Onion services that requires client authorization won't load on Tor Browser mobile

DOCS:
* [Relay requirements](https://community.torproject.org/relay/relays-requirements/) - A RPI can handle to be a relay, but not if adding more services running like we do on Blitz.

More will be added as I remember them or encounter similer

GiovanniZotta, 2021-06-28:
I was thinking that, since I'm running a BTC node on a rasPi, I might as well run a Tor relay to improve the Tor network. I understand that this is not strictly related to the raspiblitz project, but I think it would have been a cool feature if people wanted to support the Tor network.

After having a discussion with a person in the Telegram group, I've come to the personal conclusion that running a relay as an **exit node at home** might not be the best idea.

Exit nodes expose themselves on a public list, and those IPs usually meet lots of captchas and sometimes are blacklisted to avoid various kinds of attacks. This is inconvenient for the average user, but you might still want to run an exit node on a VPS provided by some cloud provider (although this is out of the scope of raspiblitz). That person suggested me that an alternative was to instead run a Tor **bridge**, which is stealthier and can still help the Tor network. 

I'll try to look around to understand if there are problems with that in terms of security/privacy/performance and I'll update here.

nyxnor, 2021-06-28:
Giovanni, Im sourcing some bridges scripts from https://github.com/radio24/TorBox, check them out.
I'm removing the (bridge) relay part, on RaspiBlitz will only be available client side to connect to bridges.
If you ask, so I can just clone TorBox on Blitz and get it going? No, different use cases, and some scripts need to be modified, since they use PasswordAuthentication and we use CookieAuthentication, regarding new_ident.sh.
Other torrc lines also could interfere with normal Blitz functionality, did not test that.