raspiblitz/raspiblitz #2384: RTL update to v0.11.0 for LND and CLN

openoms, 2021-06-28:
```
$ config.scripts/bonus.rtl.sh 

# config script to switch the RideTheLightning WebGUI on, off or update

# bonus.rtl.sh [on|off|menu] <lnd|cln> <testnet|signet>
# sets up lnd on mainnet by default
# able to run intances for lnd and cln parallel
# lnd mainnet and testnet can run parallel
# cln can only have one network active at a time

# bonus.rtl.sh [update<commit>|config]
# installs the version v0.11.0 by default
```

timalpha, 2021-07-07:
mv: cannot move '/home/admin/RTL-Config.json' to '/home/rtl/RTL/RTL-Config.json': Permission denied

Line 171: sudo -u rtl mv /home/admin/RTL-Config.json /home/rtl/${netprefix}RTL/

I think it should be just: sudo mv /home/admin/RTL-Config.json /home/rtl/${netprefix}RTL/

openoms, 2021-07-07:
yes, thanks. Fixed in here: https://github.com/rootzoll/raspiblitz/pull/2352/commits/80a0e7a77eaa24c5b3467b05b49031f2e5a26a31

waiting to be merged and reach the dev branch as part of the refactoring for v1.7.1

rootzoll, 2021-10-09:
new version looks good. closing issue for release.

closed 2021-10-09 by rootzoll