raspiblitz/raspiblitz #2386: Changing the ALIAS in lnd.conf does not change name when using punctuations

litux, 2021-06-29:
Changed Raspiblitz Node Alias in lnd.conf and restarted the lnd service 
Want to use punctuations there (Like this: PLEBNET.org-NAME) The punctuation and - do not get picked up unfortunately 
How to solve this?

openoms, 2021-06-29:
edit the lnd.conf directly. from the SYSTEM meu or:

`sudo nano /mnt/hdd/lnd/lnd/conf`

litux, 2021-06-30:
I did this initially directly on the command line and restarted lnd Took some time to pick this up as it seems, can see it now in the emboss.space site In the terminal "raspiblitz" Menu it isn't picked up though, the Status screen title still without punctuation and -

openoms, 2021-06-30:
special characters are filtered to not cause problems in the scripts. See: https://xkcd.com/327/

Can try editing the hostname in  `/mnt/hdd/raspiblitz.conf`. Put the string between single quotes: '

litux, 2021-06-30:
That did the trick in the terminal, tnx

closed 2021-06-30 by litux