raspiblitz/raspiblitz #2387: Dns queries

LAvULDER, 2021-06-30:
I migrated recently from umbrel to raspiblitz.
In the last 24h it made over 18300 dns queries. It queried for v4.ipv6-test.com and mempool.space, ~9000 times both of them. 
See [screenshots (imgur)](https://imgur.com/a/ouDgwEE) . The domains are not blocked by pihole and are resolved correctly by unbound.
I uninstalled mempool, then the dns queries for mempool.space stopped. Made a reboot, installed it, same "problem" as before.
I am running v1.7.0 behind Tor

rootzoll, 2021-07-01:
We had similar reports before - repopned the old issue #1909. Will look into the calls of `v4.ipv6-test.com` that comes from our raspiblitz scripts.

About the `mempool.space` calls ... @LAvULDER please report over at https://github.com/mempool/mempool about it because that would need to be fixed on their side. If the fix & release a new version would be great to have it in the next RaspiBlitz release.

LAvULDER, 2021-07-01:
Thanks,  will open an issue over at mempool

fanaticalfishing, 2022-01-27:
Raspiblitz is making  thousands of DNS queries for v4.ipv6-test.com.

I looks like the problem is here.
 raspiblitz/home.admin/config.scripts/internet.sh 

If IP6 is enabled or not. This script will try to ping v4.ipv6-test.com

  ###########################################
  # Global IP
  # the public IP that can be detected from outside
  globalIP=""
  echo "# getting public IP from third party service"
  if [ "${ipv6}" == "on" ]; then
    globalIP=$(curl -s -f -S -m 5 http://v6.ipv6-test.com/api/myip.php 2>/dev/null)
  else
    globalIP=$(curl -s -f -S -m 5 http://v4.ipv6-test.com/api/myip.php 2>/dev/null)
  fi
  echo "##  curl returned:  ${globalIP}"
  echo "##  curl exit code: ${?}"

mitohund, 2022-03-02:
Just wanted to confirm that my Raspiblitz 1.7.2 queries both mempool.space and v4.ipv6-test.com 4-6 times per minute each.

In fact, the Raspiblitz generates over 43% of all DNS requests of my entire home network consisting of over 20 internet-enabled devices.

rootzoll, 2022-03-08:
Its on the list to work on for v1.8 release. Can you confirm that you run your nodes on clearnet? Or do you run with Tor on?

mitohund, 2022-03-08:
Tor is on.

rootzoll, 2022-03-08:
Hmm OK thx  - on my pihole test for 1.7.2 the DNS queries looked moderate when running Tor - have to recheck.

flyerbear, 2022-05-08:
I wanted to update that I see the same issue on my pihole with both `v4.ipv6-test.com` and `mempool.space`. I reached out to the mempool.space guys on their keybase support chat and after some back and forth, they believe the issue is something with the raspiblitz code rather than their own. Just an additional data point and info.

mitohund, 2022-06-09:
The Raspiblitz is now causing over 75% of all DNS queries in my network, which as mentioned consists of over 20 Internet-enabled devices. I counted 27 requests in the last minute:
- mempool.space has disappeared
- a new query has shown up: <changing-IP>.in-addr.arpa. What is this for?
- the rest is still for v4.ipv6-test.com

I'm still on 1.7.2. I believe the only thing I changed was that I installed BitcoinMinds. Uninstalled it now to see if this would change anything. It seems like it has not.

rootzoll, 2022-06-30:
v.1.8.0 should also ready have some more reduction (especially when electrs is isntalled)

closed 2022-12-15 by rootzoll

mitohund, 2022-12-19:
Just wanted to confirm that DNS queries have mostly disappeared for me (1.8.0c, electrs 0.9.10). Thanks for solving this!