raspiblitz/raspiblitz #2389: Redesign OS builds

AaronDewes, 2021-07-01:
This is a suggestion for a feature I'd work on, but first, I want to get your feedback about this.

The current build system which "injects" RaspiBlitz into existing images isn't a great solution in my opinion, because

- These images means you have to trust third parties for building them
- Some stuff needs to be removed that is in the prebuilt image
- It is rather complicated to build new images

My proposal would be based on `vmdb2` and build a pure Debian/Ubuntu (Both work, what would you prefer?) system from scratch, in a secure, verifiable way with automated builds. For every currently supported platform, configuration could be generated. I created an automated config generator for vmdb2, which can be used for this: https://github.com/AaronDewes/vmdb-config

rootzoll, 2021-07-01:
Thanks for the idea - like the verifiable part :) At the moment we build on the official RaspberryPi image because its most optimized for the RaspberryPi hardware. @AaronDewes when you build from scratch will that still be possible?

 @openoms did the most testing on alternative OS platforms - what is your take on this?

AaronDewes, 2021-07-01:
Raspberry Pi OS' build system is fully open source, so we can do any optimization they do, but the build from Scratch would use Debian, not RPi OS, which means it's a bit different, but we can fix most issues. It needs one workaround though, because only RPi OS supports the WiFi chip of the Pi 400 (Pi 4 is supported though).

Even if you want to build on the official image, forking their build infrastructure and extending it for Blitz might be better: https://github.com/RPi-Distro/pi-gen

rootzoll, 2021-07-01:
I would be open for this if you @AaronDewes want to contribute on that angle - having a clean build based on debian would be great :) @openoms & @frennkie would you ACK going that way?

openoms, 2021-07-01:
concept ACK for sure, I am fully supporting this.

I prefer Debian because it is upstream to Ubuntu.

The Raspberry OS images are built on Debian adding the propriatery packages for the specific board support.
Similar is done with raspi.debian.net with minimal bloat. The pi-gen should be a good start.

Changing the base image again for v1.8 could be a good target if tests go well.

rootzoll, 2021-07-02:
@AaronDewes so green light for your proposal :) Keep us updated on your progress

AaronDewes, 2021-12-01:
I'm working on https://github.com/AaronDewes/blitz-gen to automate the build completely using GitHub actions, based on Raspbian. #2754 will help with that project, but I can't automate it on Github until they support Ubuntu 22.04.

cstenglein, 2023-06-07:
There doesn't seem to be much activity here => closing for now

closed 2023-06-07 by cstenglein