raspiblitz/raspiblitz #2390: Unit bitcoind.service could not be found.

Tyson-Smiter, 2021-07-01:
Finally finished downloading the blockchain from a trusted laptop node. Raspiblitz initially said that bitcoind service was running, but after an hour wait, bitcoind service is NOT running.

Should I try to restart everything from scratch again or is there a way I can salvage/debug this?

Link to error logs:
https://termbin.com/bkpg

[image]

openoms, 2021-07-01:
at this point best to reflash the Sdcard. If you have the chain on the disk already it will be picked up during the new new bitcoin node setup.

Tyson-Smiter, 2021-07-01:
Thank you. Reflashing did the trick

closed 2021-07-01 by Tyson-Smiter