raspiblitz/raspiblitz #2394: Initial setup of custom build failing with "no value for network in /home/admin/raspiblitz.info"

raucao, 2021-07-02:
I followed [the instructions for building the SD card image](https://github.com/rootzoll/raspiblitz#build-the-sd-card-image), and the initial setup script is failing with the following error:

```
FAIL: Something is wrong. There is no value for network in /home/admin/raspiblitz.info.
Should be at least default value. EXIT
```

The config key does not exist in that file, indeed. But there's no info I could find in this repo about what the value should be. I also couldn't find any default being set in the build script.

openoms, 2021-07-03:
the build_sdcard.sh defaults to the `dev` branch which is behind.

Would suggest to reflash with the 64bit RAspberry OS image and run the build with the options:

`sudo bash build_sdcard.sh false false rootzoll v1.7`

raucao, 2021-07-03:
I tried that, but it's yielding the same result:

```
admin@raspberrypi:~ $ sudo cat /home/pi/.bash_history |grep build
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh
sudo bash build_sdcard.sh false false rootzoll v1.7
admin@raspberrypi:~ $ ./10setupBlitz.sh 
checking setup script
network()
chain()
setupStep()
FAIL: Something is wrong. There is no value for network in /home/admin/raspiblitz.info.
Should be at least default value. EXIT
```

openoms, 2021-07-03:
Not sure what is  happening there.

Flash the Raspberry OS fresh then run in pi user:
```
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh
sudo bash build_sdcard.sh false false rootzoll v1.7
```
The setup process and menu will start automatically once you reboot and login with the admin user.

raucao, 2021-07-03:
I added the grep'ed bash history to demonstrate that this is exactly what I did.

raucao, 2021-07-03:
The code here definitely does not write a `network` config to that info file:

https://github.com/rootzoll/raspiblitz/blob/v1.7/build_sdcard.sh#L555-L558

But the same 1.7 branch's setup script also definitely requires it:

https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/10setupBlitz.sh#L17-L21

openoms, 2021-07-03:
the info file is filled by the `_bootstrap.sh` run by `/etc/systemd/system/bootstrap.service`

Check 
`sudo systemctl status bootstrap.service`

Restart:
`sudo systemctl restart bootstrap.service`

Or restart the whole node again.

Would recommend to rebuild if something has gone wrong in the start to have a clean state.

raucao, 2021-07-03:
Aha, now we're getting closer! The bootstrap script seems to be hanging at `btrfs filesystem show`.

I checked, and the HDD is still unformatted. `/var/log/bootstrap.log` is empty. As the SD card's filesystem is EXT4, I guess it shouldn't be running any btrfs checks to begin with, but I verified that running that command doesn't work (but also doesn't fail, as it just hangs).

The `btrfs filesystem show` commands (multiple processes) are coming from `/home/admin/config.scripts/blitz.datadrive.sh`. So I would assume they're supposed to fail instead of hang. I'll check the disk to see if I can find out more...

raucao, 2021-07-03:
Actually, wouldn't it be useful if the setup script checked that the bootstrap was successful? If it had told me that there's a bootstrap and that it didn't complete, then everything would have been clear to me from the start...

closed 2021-07-03 by raucao