raspiblitz/raspiblitz #2395: GPL verbatim mentioning Raspiblitz developers

nyxnor, 2021-07-03:
Im including GPL code from TorBox so I need to use the verbatim inside the files.
Can I include Raspiblitz Developers in a GPL verbatim as Blitz currently uses MIT?
[image]

rootzoll, 2021-07-03:
Where are you including GPL code? Depending on the case this could lead because of GPLs copyleft for RaspiBlitz to change from MIT to GPL for everything. Can you give more details?

nyxnor, 2021-07-03:
Yes, I've been working on facilitating the use of bridges with Tor and @openoms pointed out to https://github.com/radio24/TorBox
They have been tested through time, users, usability. The scripts they use for managing bridges are much more complete and help a lot to configure bridges.
It is only one part of the Tor menu, but it is 10 files being sourced from there, 8 of them are being maintained almost intact as they work in any system. The others I need to rewrite. Also, more function were integrated to already existent Blitz scripts that can help it become more readable and handling errors.
For now, it has 1800 lines of code, I've read all the lines, just need to finish my revision.
I abandoned the way I tried to add bridges with the menu because this was already contained in their code to add bridges manually, but they also acquire directly from the database, use different puggable transports.

rootzoll, 2021-07-03:
Sounds reasonable. @openoms & @frennkie can you take a look because this would trigger a license change for the RaspiBlitz Project as mentioned here: https://github.com/rootzoll/raspiblitz/issues/70#issuecomment-864338485

openoms, 2021-07-03:
I am happy with the changes as needed. Makes sense to keep the MIT licence as long as possible.

Just for the record JoinMarket is GPL too, but their code is not used either only installed as a service (same in JoininBox).

rootzoll, 2021-07-19:
Hmmm thanks to @nyxnor i checked again if we need to change the whole project license if we add single scripts under GPL .. if we qualify for an "aggregate" (what seems quite likely - because we are just bundling those scripts) it may even be possible with GPLv2 & GPLv3 - see this article: https://www.zdnet.com/article/gplv3-myth2-you-cant-mix-gpl-software-with-other-software/

nyxnor, 2021-07-22:
Got it, closing.

closed 2021-07-22 by nyxnor