raspiblitz/raspiblitz #2398: Repeated password C loop, unable to connect to bitcoind: -28: Loading block index

stevb, 2021-07-04:
I've tried searching and it seems there are others with similar issues, but I haven't been able to use their posts to solve my own issue. Using a Pi4 with 4GB, samsung T5 1TB SSD, official raspi 3A PSU, 32GB sandisk microSD. Took about 10+ days to DL the blockchain and associated lightning bits, so hoping I don't have to DL again.

The output for ./XXdebugLogs.sh is visible at this [link](https://termbin.com/jez9c)

The display on the Pi says bitcoind is not running, loading block index. In the error logs, I'm suspicious of the following line, but I don't know how to remedy my situation:

LTND: unable to create chain control: unable to connect to bitcoind: -28: Loading block index...

Thanks for your help and insight!

openoms, 2021-07-05:
Your node is reindexing (rolling forward). Will work when gets to the tip of the chain.

stevb, 2021-07-05:
Thanks for your reply; In the meantime, I've done a proper shutdown via ./XXshutdown.sh to the white screen and then pull the plug. Is it simply a matter of waiting for this to catch up to the latest blocks? I've noticed this morning that the `NOT RUNNING' screen now simply says `LND IS LOCKED' or something to that effect. Will try restart now.

stevb, 2021-07-05:
Ok, seems to be up and running now. Wasn't really clear that I simply needed to keep waiting. Thanks for the help!

closed 2021-07-05 by stevb