raspiblitz/raspiblitz #2402: Fail to activate JoinMarket

cricktor, 2021-07-05:
I run RaspiBlitz v1.7.0 on a Pi 4B with 8GB RAM and a 1TB Sandisk SSD since maybe two weeks.
Prior to activate JoinMarket from menu, the following modules are running fine and synchronized behind TOR:
bitcoind, LND, Electrum Rust Server, BTC RPC Explorer, Mempool Space, RTL Webinterfache

I selected JoinMarket from "Services" menu and it runs into a fail message box:
JoinMarket installation is cancelled
Try again from the menu or install from the terminal with:
sudo /home/admin/config.scripts/bonus.joinmarket.sh on

I tried this command from terminal to preserve the printout and got the following output:
```
admin@raspberrypi:~ $ sudo /home/admin/config.scripts/bonus.joinmarket.sh on
*** INSTALL JOINMARKET ***
# OK, running behind Tor
Multiwallet is active and wallet.dat is used.
The bitcoin core wallet is already on
# cleaning before install
# add the 'joinmarket' user
Lege Benutzer »joinmarket« an ...
Lege neue Gruppe »joinmarket« (1008) an ...
Lege neuen Benutzer »joinmarket« (1008) mit Gruppe »joinmarket« an ...
Erstelle Home-Verzeichnis »/home/joinmarket« ...
Kopiere Dateien aus »/etc/skel« ...
# setting PASSWORD_B as the password for the 'joinmarket' user
Füge Benutzer »joinmarket« der Gruppe »sudo« hinzu ...
Benutzer joinmarket wird zur Gruppe sudo hinzugefügt.
Fertig.
joinmarket ALL=(ALL) NOPASSWD:ALL
# installing ARM specific dependencies to run the QT GUI
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
libjpeg-dev ist schon die neueste Version (1:1.5.2-2+deb10u1).
libltdl-dev ist schon die neueste Version (2.4.6-9).
python3-pyqt5 ist schon die neueste Version (5.11.3+dfsg-1+b3).
python3-pyside2.qtcore ist schon die neueste Version (5.11.2-3).
python3-pyside2.qtgui ist schon die neueste Version (5.11.2-3).
python3-pyside2.qtwidgets ist schon die neueste Version (5.11.2-3).
zlib1g-dev ist schon die neueste Version (1:1.2.11.dfsg-1).
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  libdw1
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
automake ist schon die neueste Version (1:1.16.1-4).
build-essential ist schon die neueste Version (12.6).
libgmp-dev ist schon die neueste Version (2:6.1.2+dfsg-4).
libffi-dev ist schon die neueste Version (3.2.1-9).
pkg-config ist schon die neueste Version (0.29-6).
python3-dev ist schon die neueste Version (3.7.3-1).
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  libdw1
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting libtool
  Downloading https://files.pythonhosted.org/packages/bf/05/35407634ae084bec879ee7541137748aeda91a3ed504a1a94d8ed00f49be/libtool-1.1.0-py3-none-any.whl
Requirement already satisfied: asn1crypto in /usr/lib/python3/dist-packages (0.24.0)
Collecting cffi
  Downloading https://files.pythonhosted.org/packages/a8/20/025f59f929bbcaa579704f443a438135918484fffaacfaddba776b374563/cffi-1.14.5.tar.gz (475kB)
    100% |████████████████████████████████| 481kB 972kB/s
Collecting pycparser
  Downloading https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl (112kB)
    100% |████████████████████████████████| 112kB 2.3MB/s
Collecting coincurve
  Downloading https://files.pythonhosted.org/packages/65/31/9f13921cc7ca27059af4def1a9d1277a6a252dea967b62f9ea05cc2a4115/coincurve-15.0.0.tar.gz (219kB)
    100% |████████████████████████████████| 225kB 1.9MB/s
  Installing build dependencies ... done
[... 92 lines cut ...]
cp: -r wurde nicht angegeben, daher wird das Verzeichnis '/home/joinmarket/joininbox/scripts/standalone' ausgelassen
runBehindTor=off
# setting the autostart script for joinmarket

# automatically start startup.sh for joinmarket unless
# when running in a tmux session
if [ -z "$TMUX" ]; then
  /home/joinmarket/startup.sh
fi
# always activate jmvenv with PySide2 and cd to scripts'
. /home/joinmarket/joinmarket-clientserver/jmvenv/bin/activate
/home/joinmarket/joinmarket-clientserver/jmvenv/bin/python -c "import PySide2"
cd /home/joinmarket/joinmarket-clientserver/scripts/
# shortcut commands
source /home/joinmarket/_commands.sh
# automatically start main menu for joinmarket unless
# when running in a tmux session
if [ -z "$TMUX" ]; then
  /home/joinmarket/menu.sh
fi

 Failed to install JoinMarket
```

Not sure if this is caused by the missing command 'virtualenv' and I'm not so familiar with the environment to know why this is missing. If you need anything more, don't hesitate to request it from me (and apologies if I behave like a total Github noob).

openoms, 2021-07-05:
Thanks for reporting. Well spotted. Try:
`sudo apt install virtualenv`

before attempting to install again.

Can clean up first with:
`sudo /home/admin/config.scripts/bonus.joinmarket.sh off`
or
`sudo userdel -rf joinmarket`

Let us know if this solves the issue and will check for that dependency,

cricktor, 2021-07-05:
Thank you, after installing `virtualenv` the installation script does a lot of more stuff; libsodium is compiled fine and passes all tests.
The cleanup with sudo `/home/admin/config.scripts/bonus.joinmarket.sh off` just responds with that JoinMarket isn't installed and of course the `sudo userdel -rf joinmarket` deletes the user and home directory.

Still some red error logging and yellow warning, not sure if something's seriously broken. I will show as much of the logging and flag the colored sections below.
```
~/joinmarket-clientserver/deps ~/joinmarket-clientserver
~/joinmarket-clientserver
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/joinmarket/joinmarket-clientserver/jmbase (from -r requirements/base.txt (line 1))
Obtaining file:///home/joinmarket/joinmarket-clientserver/jmbitcoin (from -r requirements/base.txt (line 2))
Obtaining file:///home/joinmarket/joinmarket-clientserver/jmclient (from -r requirements/base.txt (line 3))
Obtaining file:///home/joinmarket/joinmarket-clientserver/jmdaemon (from -r requirements/base.txt (line 4))
Collecting twisted==20.3.0
  Downloading Twisted-20.3.0.tar.bz2 (3.1 MB)
     |████████████████████████████████| 3.1 MB 4.2 MB/s
Collecting service-identity
  Downloading https://www.piwheels.org/simple/service-identity/service_identity-21.1.0-py2.py3-none-any.whl (12 kB)
Collecting chromalog==1.0.5
  Downloading https://www.piwheels.org/simple/chromalog/chromalog-1.0.5-py2.py3-none-any.whl (10 kB)
Requirement already satisfied: coincurve in /home/joinmarket/.local/lib/python3.7/site-packages (from joinmarketbitcoin==0.8.2->-r requirements/base.txt (line 2)) (15.0.0)
Collecting python-bitcointx>=1.1.1.post0
  Downloading python_bitcointx-1.1.1.post2-py3-none-any.whl (189 kB)
     |████████████████████████████████| 189 kB 7.7 MB/s
Collecting pyaes
  Downloading https://www.piwheels.org/simple/pyaes/pyaes-1.6.1-py3-none-any.whl (26 kB)
Collecting urldecode
  Downloading https://www.piwheels.org/simple/urldecode/urldecode-0.1-py3-none-any.whl (2.3 kB)
Collecting mnemonic
  Downloading https://www.piwheels.org/simple/mnemonic/mnemonic-0.19-py2.py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB 1.4 MB/s
Collecting argon2_cffi
  Downloading argon2-cffi-20.1.0.tar.gz (1.8 MB)
     |████████████████████████████████| 1.8 MB 7.7 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
```
#### red colored section
```
    ERROR: Command errored out with exit status 1:
     command: /home/joinmarket/joinmarket-clientserver/jmvenv/bin/python3 /home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp1l6jsws4
         cwd: /tmp/pip-install-kjwep4lg/argon2-cffi_2fa9eced09964bb3adab04acc582d92c
    Complete output (34 lines):
    Traceback (most recent call last):
      File "/home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 280, in <module>
        main()
      File "/home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 140, in prepare_metadata_for_build_wheel
        self.run_setup()
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 126, in run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 387, in <module>
        **keywords_with_side_effects(sys.argv)
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.7/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 447, in __init__
        k: v for k, v in attrs.items()
      File "/usr/lib/python3.7/distutils/dist.py", line 292, in __init__
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 739, in finalize_options
        ep.load()(self, ep.name, value)
      File "/tmp/pip-build-env-s_3925dt/overlay/lib/python3.7/site-packages/cffi/setuptools_ext.py", line 219, in cffi_modules
        add_cffi_module(dist, cffi_module)
      File "/tmp/pip-build-env-s_3925dt/overlay/lib/python3.7/site-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
        execfile(build_file_name, mod_vars)
[... 232 lines cut ...]
# setting the autostart script for joinmarket

# automatically start startup.sh for joinmarket unless
# when running in a tmux session
if [ -z "$TMUX" ]; then
  /home/joinmarket/startup.sh
fi
# always activate jmvenv with PySide2 and cd to scripts'
. /home/joinmarket/joinmarket-clientserver/jmvenv/bin/activate
/home/joinmarket/joinmarket-clientserver/jmvenv/bin/python -c "import PySide2"
cd /home/joinmarket/joinmarket-clientserver/scripts/
# shortcut commands
source /home/joinmarket/_commands.sh
# automatically start main menu for joinmarket unless
# when running in a tmux session
if [ -z "$TMUX" ]; then
  /home/joinmarket/menu.sh
fi

Start to use by logging in to the 'joinmarket' user with:
sudo su joinmarket

If logging in directly via ssh the password is the PASSWORD_B
```
I will have to do some testing if JoinMarket works OK and as expected or if something is broken due to the errors that occurred.

openoms, 2021-07-06:
This looks good. `cffi` is attempted to be installed using different versions until succeeds. This is the behaviour of the original install script and leads to no problems (apart from ugly colours).

JoinMarket should work as intended.

The best next step is to update Joininbox (just pulls the github repo) then update JoinMarket to the latest v0.8.3 version.
Also recommended to reset the joinamrket.cfg after update to correct the new Tor v3 IRC server addresses.

openoms, 2022-01-06:
closing for inactivity

closed 2022-01-06 by openoms