raspiblitz/raspiblitz #2405: [Discussion] disk mounted with `noexec` - is it useful?

openoms, 2021-07-06:
The disk is currently mounted with `noexec` in the `/etc/fstab` so nothing on it can be executed.

For now we need to store any extensions (https://github.com/rootzoll/raspiblitz/pull/2404) and C-lightning plugins on the SDcard and redownload them after SDcard changes unless this changes.

`noexec` is applied in the RaspiBolt guide (https://stadicus.github.io/RaspiBolt/raspibolt_20_pi.html#format-external-drive-and-mount), but does it have a real security benefit?

m00ninite, 2021-07-08:
fwiw, the `noexec` mount has screwed me up multiple times, aside from that extensions PR. I'd also love to know if there's a security benefit.

xanoni, 2021-07-14:
I'm not 100% sure if this is best practice, but why not just do 'mount -oremount,exec /xyz', execute the commands, and then lock it up again with'-oremount,noexec'?

AFAIK this works fine even for a hot system ... but maybe I just always got lucky ;-)

xanoni, 2021-08-02:
Just confirmed and this works fine on my machine:

```bash
$ mount |grep disk01
/dev/mapper/cdisk01 on /mnt/disk01 type ext4 (rw,nodev,noexec,noatime,discard)
$ sudo mount -oremount,exec /mnt/disk01/
$ mount |grep disk01
/dev/mapper/cdisk01 on /mnt/disk01 type ext4 (rw,nodev,noatime,discard)
$ sudo mount -oremount,noexec /mnt/disk01/
$ mount |grep disk01
/dev/mapper/cdisk01 on /mnt/disk01 type ext4 (rw,nodev,noexec,noatime,discard)
```

closed 2021-08-11 by openoms

chrisguida, 2025-02-01:
Yeah I don't really see the benefit here.

Seems like any malware that might get installed on the ssd can just as easily be installed on the sd card.

rootzoll, 2025-02-03:
The idea was also to enforce the seperation between data and system .. the SSD is just meant to be only for data and all system/app-executables shoule be seperated on the sd card.

> Building Rust CLN plugins is especially painful with this on.

@chrisguida can you describe how you need the SSD space for building CLN plugins

chrisguida, 2025-02-03:
@rootzoll my specific case is building Rust CLN plugins on the raspi 4B, which has all the executable stuff on the sd card, which is super slow and only 32GB. So not only do the builds take forever, but Rust build artifacts tend to take up lots of space and fill up the disk. It would be 10x better to perform these builds on the external SSD which is much larger and faster, but it's not obvious how to accomplish this in an ergonomic way, since given Blitz's CLN plugin architecture where the plugins are built in `cl-plugins-available` then symlinked in `cl-plugins-enabled`, the executable is always on the same drive where the build occurred.

The easiest way I've found to work around this is to remount the drive as `exec`.

rootzoll, 2025-02-03:
@chrisguida thx for the details. For RaspiBlitz v1.12.0 we are working on making boot possible from NVMe - see #4324 - so running & building things will get much faster by default BUT that will only be possible with RP5 & NVMe. On storage devices/partitions I would like to keep the`exec` - hope the work around in your special case on older hardware is OK for you.

chrisguida, 2025-02-03:
>On storage devices/partitions I would like to keep the exec

Can you explain the reasoning? What is the precise attack vector that is mitigated by keeping the data drive as `noexec`?

>hope the work around in your special case on older hardware is OK for you.

If you mean remounting my drive as `exec`, yes, I'm fine with that if that's the best way.

rootzoll, 2025-02-03:
> Can you explain the reasoning? What is the precise attack vector that is mitigated by keeping the data drive as noexec?

Its not about preventing attacks - its about keeping order and things seperated. Install/System drives is for executables and data/storage for the rest - thats at least the idea. Install and System can get replaced & overwritten during updates - data stays seperated and should not be mixed with parts of the system.