raspiblitz/raspiblitz #2409: Tor menus and features

nyxnor, 2021-07-06:
This is to track what remains to be done.
It is just checked if I have tested it and worked to the maximum extent.

 "=[Checks]==========================================================")
- [ ] CREDENTIALS "See services credentials (address, key, QR code)"
- [ ] NYX "Terminal status monitor for tor")
- [ ] LOGS "Show the logs")
- [ ] TORRC "Edit configuration file (torrc)"
- [ ] RESTART "Restart Tor service or other instances"
- [ ] SSH_OVER_TOR ${SSHTORSTRINGb}" SSH over Tor"
"=[Countermeasure]=================================================="
- [ ] AUTH "Request client authentication for onion service"
- [ ] RENEW_ADDRESS "Request new onion services addresses"
- [ ] VANGUARDS ${VANGUARDSSTRINGb}" protection against server location deanonymization"
- [ ] BRIDGES ${BRIDGESTRINGb}" Tor bridges with pluggable transport"
- [ ] SOURCES "Request APT over Tor and update packages"
- [ ] CHANGE_CIRCUIT "Request new circuit (signal NEWNYM)"
- [ ] CHANGE_GUARD "Delete all circuits and force change of the entry node"
- [ ] BYPASS_IDLE ${PINGb}" countermeaseure against idle feature"
- [ ] OVER_VPN ${VPNSTRINGb}" Tor over VPN"
============== "=[Informational]==================================================="
- [ ] VERSION "Tor related packages version"
- [ ] ONION "Onion routing privacy and security benefits"
- [ ] DISCLAIMERS "What Tor does not provide?"
- [ ] SUPPORT_TPO "Support the Tor Project"
- [ ] SUPPORT_TORBOX "Support TorBox"

nyxnor, 2021-07-06:
* VANGUARDS - Code implemented but needs more testing.

* CHANGE_GUARD - Code implemented but did not test thoroughly yet.

* BRIDGES - Code not finished and here is why. We are using 2 control ports with different files, it is difficult to sync the same config to two files, it could get out of sync and cause bugs.
files are `/etc/tor/torrc` and `/etc/tor/instances/lnd/torrc`.
The only viable solution I think could work is to always configure on `/etc/tor/bridges` and warn to not change bridges config inside the main torrc as it will be overwritten. The plan of using `bridges` file is to make it canonical and delete the old bridges lines from the 2 torrc with sed then cat the new bridges config to it. Sed will help to just delete matching lines so it won't delete other lines by mistake. This will take a bit more time to do.

nyxnor, 2021-07-06:
### Improvements
* Brings options for users to protect their onion service either by handling DDoS by restricting access with Client Authorization or by handling attack to deanonymize server location with Vanguards.
* You can APT over Tor for Debian and Tor Project sources (v3 addresses)  and change it with a menu click any time, can be none, one of them or both over Tor.
* Onion service address renewal in case you leak them or just want to change
* See Tor credentials such as the address, the client key and the QR to connect to it from the terminal
* Changing circuits and entry node, Should change circuits more frequently than entry node, entry node is advanced usage and not expected.
* Tor will restart just when needed, it will sighup more than restart now, having less downtime.
* Countermeasure such as Bridfges (secure and anonymous), over VPN (potentially less anonymous) and bypass idle (when there is disconnection idle feature, suitable when travelling with the node on cafeterias, airports, hotels).
* SSH over Tor warns it is for advanced users and should use it cautiously, also incentivizes to use together with Client Auth
* Tor update will still be in Menu > Update > Tor, but it will use git to upgrade to any version.
* Most of the options prompt text explaining what are their functionalities and the important things to note when using it.
* Instruction texts at the bottom for user to understand more how Tor works, what it does not help, how to support the used projects.

nyxnor, 2021-07-06:
### IMAGES

#### Main menu
[image]

#### Bridges menu
[image]

#### Menu for Auth, credentials, renewal
[image]

#### APT over Tor
[image]

#### Choose instance (depending on the option, just edit, restart, sighup the chosen instance to avoid downtime)
[image]

originaljeremy, 2021-07-07:
This is fantastic work. Thank you for doing it.

nyxnor, 2021-07-08:
## Restarting, reloading with specific ports or the whole tor

Restarting can take a long time, you can edit torrc files without needing to restart tor to apply changes, you can just reload it with the methods described below.

### Reload all tor instances
`sudo pkill -sighup tor` 
[image]

### Reload specific port with 1 command

https://stem.torproject.org/tutorials/down_the_rabbit_hole.html

#### 9051 for tor@default

`sudo -u debian-tor tor-prompt --run 'SIGNAL HUP' -i 9051`
[image]

#### 9071 for tor@lnd

`sudo -u _tor-lnd tor-prompt --run 'SIGNAL HUP' -i 9071`
[image]

### Reload specific port with 1 file and can include password

https://stem.torproject.org/faq.html#how-do-i-reload-my-torrc

* we don't use `PasswordAuthentication` in the torrc, but storing this here for the future
* if there is no password, just cookie, we can still use this, but with `controller.authenticate()` empty
```python3
#!/usr/bin/env python3

from stem import Signal
from stem.control import Controller

with Controller.from_port(port = 9051) as controller:
  controller.authenticate(password=yourPassword)
  controller.signal(Signal.HUP)
```

This was working on the pi before, now I don't know why it is not working
[image]

nyxnor, 2021-07-14:
### Ideas for future changes

This cannot be implemented now due to time limitation and complexity

* Manage client authorization, removing specific clients, giving names for them.

nyxnor, 2021-07-22:
Documentation is still missing, not changes made to the FAQ or README, but I think brief notes there would help, as text/ folder is pretty resourceful on how to use each option.

nyxnor, 2021-07-26:
Latest tag from tor git:
```
git describe --tags $(git rev-list --tags --max-count=10)
```
```
tor-0.4.6.6
tor-0.4.6.5-46-g60ffee9d82
tor-0.4.6.5-45-g01fd3ace03
tor-0.4.6.2-alpha-108-g2fff1fc835
tor-0.4.6.5-43-gbd3ed77191
tor-0.4.6.2-alpha-107-g4302c0b4a1
tor-0.4.5.1-alpha-395-ge60d14bb6b
tor-0.4.6.5-40-g2630e4eea1
tor-0.4.6.2-alpha-105-gf2647295c0
tor-0.4.5.1-alpha-394-gfce99957e2
```

rootzoll, 2021-08-17:
Its great work .. just because we have a lot of moving parts at the moment for the v1.7.1 with c-lightning and the setup-process refactor I will push this one realese up to 1.8.

nyxnor, 2021-09-14:
This is being redesigned because I did not know it was that hard to keepup with TorBox changes even being a small project, to keep getting updates, I need to do differently.

As suggested by Openoms to make a pluggable project, it will work this way:
On my repo, I use a script to

- clone TorBox
- cp the files that I need about bridges
- remove the rest
- rename the files inside the scripts
- renames the file names
- create blitz directories
- place the files in the correct blitz directory
-
- clone onion-cli, it being a submodule of this different repo
- that is all that is needed basically
-
- add some scripts to this repo for
- reloading, restarting tor
- logs
- nyx :)
- update option for this project
- change circuit
- change guard

nyxnor, 2021-09-19:
Thanks to @openoms amazing dialog colors, used the same as Joininbox .dialogrc
[image]