raspiblitz/raspiblitz #2410: bitcoind error loop

breathe08, 2021-07-06:
Ive recently migrated from umbrel to raspiblitz following the connectivity issues in version 3.13. I was unable to completely login to my node[48 hours] which had 40 active channels when i decided to migrate as a last ditch effort before a 2 week road trip with family. I performed the migration Tuesday June 29th and it finally synced yesterday July 5th. I was notified by lightning watch that my node was back online and a few peers also saw my channels were active.

While installing TH and RPC the node began throwing errors [log below]. I performed DELETE-INDEX and the next suggestion seems to be RESET-CHAIN which I would like to avoid if possible. I am again offline but it seems bitcoind related.

Most recent debug log https://termbin.com/rl0k

I appreciate any assistance and while I have little prior experience I am happy to learn. Finally, I am on vacation with my family so my time committed here is somewhat condensed. I will do my best to respond as possible. Thank you!

breathe08, 2021-07-06:
Also I noticed during restart that there is bitcoind (1) and bitcoind (2) that need to shutdown. I have not seen this before.

openoms, 2021-07-06:
Unfortunately it seems your blockchain is corrupted on the disk:
```
Unable to open file /mnt/hdd/bitcoin/blocks/blk00020.dat
```
It can be just because of a cold reset (use a UPS), but can be a sign of failing hardware as well.

You will certainly need to reindex /redowload the chain unless you can copy from an other node.

Make sure that your power connection is solid and the power supply is a quality one. I would also consider changing the disk before more valuable data (your lightning database) gets corrupted.

breathe08, 2021-07-06:
Thank you openoms. I will follow these recommendations.

closed 2021-08-17 by rootzoll