raspiblitz/raspiblitz #2411: ban aws ip's

originaljeremy, 2021-07-07:
I can't remember who initially posted this, but I just found an old snippet I saved for banning the ip range of amazon web services to protect against a possible future sybil attack on the consensus layer, and I thought it might be interesting to discuss whether something like this could be desirable for the Blitz. I imagine it as an option when building core, giving the user the ability to ignore the range when searching for peers during IBD and later. I haven't really come up with arguments against using something like this, but I am curious what everyone thinks. Sorry I forgot who deserves credit for this:

Not trying to PR this, just starting a discussion :)

curl -s 'https://ip-ranges.amazonaws.com/ip-ranges.json' | jq -rc '.prefixes[].ip_prefix | {"jsonrpc": "1.0", "id":"aws-banscript", method: "setban", "params": [., "add", 3600]}' | jq -s  | curl -s -u rpcuser:rpcpass --data-binary @/dev/stdin -H 'content-type: text/plain' 127.0.0.1:8332

tweak the script until it works first. add it to ~/bin/banips, then chmod +x ~/bin/banips, then

0 * * * * $HOME/bin/banips

to your `crontab -e`

need this in bitcoin.conf:

rpcbind=127.0.0.1
rpcallowip=127.0.0.1
rpcuser=rpcuser
rpcpassword=rpcpass

originaljeremy, 2021-07-08:
Or maybe expand it a bit and have an option to ban all known large datacenter ip ranges so you connect to a more authentic peer group.

xanoni, 2021-07-14:
Nice. Will play with that!!

a5an0, 2021-08-17:
What about all the lightning nodes running on major cloud providers (like AWS)?

closed 2021-09-16 by originaljeremy