raspiblitz/raspiblitz #2413: I2P - [ FEATURE REQUEST]

nyxnor, 2021-07-07:
https://twitter.com/jonatack/status/1366764964896075776
https://twitter.com/openoms/status/1412879980254711809

https://github.com/bitcoin/bitcoin/blob/master/doc/i2p.md
https://en.wikipedia.org/wiki/I2P

Implement I2P as an network option.

nyxnor, 2021-07-08:
I2P + Tor + IPV4/IPV6 can be activated at the same time, but this is for later, first thing is to permit Blitz be an I2P router.

Host https://geti2p.net/en/get-involved/guides/reseed

xanoni, 2021-07-14:
For what it's worth, I've been running I2P and Tor in parallel for a while and it's really smooth. By now around 1/3 of my peers connect via I2P (the rest Tor).

xanoni, 2021-08-02:
Just saw this thread and thought I'd mention that in the most recent `bitcoind` (`HEAD`) the ports of all I2P nodes specified now need to have port `0`. My node was no longer connecting to I2P nodes but changing all the ports (e.g., `addnode` lines) to `0` fixed it.

xanoni, 2021-08-02:
Oh, and if you decide to roll this out, I would suggest that you don't use the standard Raspbian `i2pd` package, because it's very very old.

It had issues with it, see here: https://github.com/PurpleI2P/i2pd/issues/1676#issuecomment-888809375

Appears to be fixed after changing to the `i2pd` project's repository.

I haven't tried the `i2p` package (the full Java implementation), so I can't say if it's better or worse.

xanoni, 2021-08-03:
I liked the comparison doc that was mentioned above (https://geti2p.net/en/comparison/tor).

These features / properties are really helpful compared to Tor (copy & paste):
* Floodfill peers ("directory servers") are varying and untrusted, rather than hardcoded
* Both TCP and UDP transports
* implicit transparent load balancing of messages across multiple peers, rather than a single path
* resilience vs. failures by running multiple tunnels in parallel, plus rotating tunnels
* Designed and optimized for hidden services, which are much faster than in Tor

Also, the roadmap stuff is cool:

- Defense vs. message count analysis by garlic wrapping multiple messages
- Defense vs. long term intersection by adding delays at various hops (where the delays are not discernible by other hops)
- Various mixing strategies at the tunnel level (e.g. create a tunnel that will handle 500 messages / minute, where the endpoint will inject dummy messages if there are insufficient messages, etc)

nyxnor, 2021-08-03:
@xanoni Can you make a hidden service script for I2P like the one we have for Tor?
https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/internet.tor.sh

This guide is not complete but you can get help https://github.com/eyedeekay/PLEASE-Write-Better-Hidden-Service-Setup-Guides

xanoni, 2021-08-04:
> @xanoni Can you make a hidden service script for I2P like the one we have for Tor?

Yeah that sounds like a cool project. Would love to work on this and realize that I need a RaspiBlitz, haha. Guess I need to buy yet another Pi.

nyxnor, 2021-09-06:
**Some valuable instructions by [Eyedeekay](https://github.com/eyedeekay)**

Re: i2p.hidden-service.sh there are a number of approaches you could take to making such a thing, which docs you will need will depend on which approach you take.

The easiest way will be to leverage the i2ptunnel.config.d directory. You could write your script in such a way that it contains a template of an i2ptunnel.config file. By passing arguments to the script, you can affect the config file the template will output. By generating and placing an i2ptunnel.config file in the i2ptunnel.config.d directory you configure a new I2P tunnel. Then you just have to restart i2ptunnel so that it picks up the new config file. All told this is probably less than 75 lines of shell script, but restarting i2ptunnel to load the config is not very convenient. The most important doc to accomplish this is the i2ptunnel.config specification: https://geti2p.net/spec/configuration#i2ptunnel-i2ptunnel-config On the other hand, I2PTunnel itself is actually a client application of the router, so it's possible to run multiple instances of I2Ptunnel at the same time. Your shell script could generate the config file and start a whole new instance of i2ptunnel using just the newly-generated config file. The client configuration https://geti2p.net/spec/configuration#clients-clients-config of the i2ptunnel application will give you the details of how to launch it, but ultimately it will be something like `java -cp /path/to/i2p/jars -jar /path/to/i2p/jars/i2ptunnel.jar /path/to/i2p/configs/new.config`.

The most convenient way for end users will be to leverage the SAM API. You could write your script to dynamically create tunnels by setting up SAM sockets. In this case your script would have to accept and configure the I2CP and Tunnel options for the tunnels by passing them to the SAM API during the SESSION CREATE operation. Additionally, for anything you need to persist an identity for, you will need to write down the keys that SAM replies with when you complete a SESSION CREATE operation. For more information about SAM: https://geti2p.net/en/docs/api/samv3 Such a tool exists, but it is in Go and not a shell script, https://github.com/eyedeekay/sam-forwarder.

In my opinion, either of these ways would be an acceptable approach to setting up an i2p.hidden-service.sh. **However,** there is a third option, maybe the easiest of all. That is the `i2p-zero` I2P distribution, which operates a controller on TCP port 8050. You can set up and tear down tunnels on an i2p-zero router using an existing script called `tunnel-control.sh`. You can find the documentation for `tunnel-control.sh` here https://github.com/i2p-zero/i2p-zero#tunnel-control.

I'll eventually get the rest of that guide written. It's very hard to write the guide in a positive way which doesn't involve chewing out a bunch of Medium bloggers, which isn't productive. It may be a while, it's probably going to end up around 12-15 pages when it's done.

nyxnor:
> Hi Mr Eyedeeekay,
> > I want to become more familiar with I2P, but as you know, the documentation is not good: https://github.com/eyedeekay/PLEASE-Write-Better-Hidden-Service-Setup-Guides
> I want to make a i2p.hidden-service.sh to generate at services at will with specified port, can you point me to the docs?
> If you finish the above repo, would be a great pleasure to spread the word about I2P and to get more people to be familiarized and manage their servers by themselves, just reading guides without asking support.
> > Where can I find your personal PGP? https://geti2p.net/en/contact
> > nyxnor
>

nyxnor, 2021-09-14:
Scratch installation for debian

debian package form debian.org
```shell
sudo apt install -y i2pd

sudo -u i2psvc i2pd --help
```

deb package from deb.i2p2.de
```shell
printf "deb https://deb.i2p2.de/ $(lsb_release -sc) main\ndeb https://deb.i2p2.de/ $(lsb_release -sc) main\n" | sudo tee /etc/apt/sources.list.d/i2p.list

curl -sS https://geti2p.net/_static/i2p-debian-repo.key.asc | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/i2p-debian-repo-keyring.gpg >/dev/null

sudo apt update -y

sudo apt install -y i2p
```

Jar file
```shell
wget -q --show-progress http://geti2p.net/en/download/1.5.0/clearnet/https/download.i2p2.de/i2pinstall_1.5.0.jar/download

java -jar i2pinstall_1.5.0.jar
```

xanoni, 2021-09-17:
Personally I prefer the C++ implementation (i2pd) .... seems leaner and it supports UDP tunnels besides TCP.

Here is how to get the repo with the latest:

https://github.com/xanoni/RaspberryPi-Setup-Scripts/blob/8c58a2645b7d6b7b8548fee9938a9f275ab4e186/scripts/add-critical-repos.sh#L23-L27

openoms, 2021-09-21:
The TrueNAS node guide has been updated to use I2P:
https://github.com/seth586/guides/blob/master/FreeNAS/bitcoin/freenas_3_tor.md#install--configure-i2pd

Still need to compile Bitcoin Core 22.0 from source on FreeBSD so my setup is not complete yet, but looking good:

[image]

openoms, 2021-12-02:
Good related discussion here: https://github.com/JoinMarket-Org/joinmarket-clientserver/pull/1038#issuecomment-945026023
Nice read: https://i2p.medium.com/20-years-of-privacy-a-brief-history-of-i2p-8f62af87ee87

openoms, 2022-01-16:
The manual process:

```
# Add repo for the latest version
# i2pd — https://repo.i2pd.xyz/.help/readme.txt
wget -q -O - https://repo.i2pd.xyz/.help/add_repo | sudo bash -s -
sudo apt-get update

# install and start i2p
sudo apt-get install i2pd
sudo systemctl enable i2pd
sudo systemctl start i2pd
```

Add to the bitcoin.conf:
```
# check for: onlynet=onion
debug=tor
debug=i2p 
i2psam=127.0.0.1:7656
i2pacceptincoming=1 
# do not add: onlynet=i2p  
```
Restart bitcoind:
```
sudo systemctl restart bitcoind
```
Monitor:
```
watch -t bitcoin-cli -netinfo 4

sudo tail -n 100 -f /mnt/hdd/bitcoin/debug.log | grep i2p
```

Can try adding nodes from this list: 
https://github.com/bitcoin/bitcoin/blob/807169e10b4a18324356ed6ee4d69587b96a7c70/contrib/seeds/nodes_main.txt#L680
Example:
```
bitcoin-cli addnode "bitcornrd36coazsbzsz4pdebyzvaplmsalq4kpoljmn6cg6x5zq.b32.i2p:0" "onetry"
```

agates, 2022-03-21:
How far along is this?  Need help, research, testing, code contribution?

nyxnor, 2022-03-22:
Research was done, missing code contribution if you want to do it, no one has started.

Some milestones:
- script that install, update and  remove i2p
  - needs to handle the bitcoin.conf
  - needs to be on the menu to be activated, deactivated, updated
- i2p only mode
- i2p combined with tor and ipv4 (public ip)

Later, if possible, i2p garlic hidden services for all services.

----

Podcast with i2p contributors:
https://optoutkoplzfgs7wl3gkg5nmtrrs7ki6ljcguf7c4w7rdsrtozlghxad.onion/episodes/s2e10-i2p/

agates, 2022-03-22:
Does lnd/c-lightning need built in i2p support to talk to peers?  I'm guessing getting Bitcoin i2p set up is step 1?

nyxnor, 2022-03-23:
> Does lnd/c-lightning need built in i2p support to talk to peers? 

Probably would need and never heard of someone trying it out.

>  I'm guessing getting Bitcoin i2p set up is step 1?

Yes.

> I can see about getting lnd to support it if that is indeed a requirement.

This can be done later, first need to organize the bitcoin.conf for i2p on installation, deinstallation and update peers maybe.

kristapsk, 2022-08-16:
Here's not yet merged I2P bonus guide for RaspiBolt, you could take it as a basis probably - https://github.com/raspibolt/raspibolt/pull/1064.

kristapsk, 2022-08-16:
> Does lnd/c-lightning need built in i2p support to talk to peers?

No, they do not and probably never will. I2P has even bigger latency issues than Tor. It's ok for onchain Bitcoin, not so with Lightning Network.

But with I2P added at least your bitcoind instance will keep running and syncing even if Tor goes down for some time and CLN/LND instances with it.

wchancao, 2022-09-26:
> 

After following this steps, i get this messages in bitcoin logs, and not getting i2p peers

```
I2P: Creating SAM session with 127.0.0.1:7656
I2P: Error listening: Receive timeout (received 0 bytes without terminator before that)
I2P: Control socket error: not connected
```

wchancao, 2022-09-26:
> @wchancao for amd64 need to download the binary:
> 
> ```
> wget https://github.com/PurpleI2P/i2pd/releases/download/2.43.0/i2pd_2.43.0-1bullseye1_amd64.deb
> 
> # verify
> wget https://github.com/PurpleI2P/i2pd/releases/download/2.43.0/SHA512SUMS
> wget https://github.com/PurpleI2P/i2pd/releases/download/2.43.0/SHA512SUMS.asc
> curl https://repo.i2pd.xyz/r4sas.gpg | gpg --import
> gpg --verify SHA512SUMS.asc
> 
> # install
> sudo dpkg -i --force-confnew i2pd_2.43.0-1bullseye1_amd64.deb
> ```

It works! Thanks!!

rootzoll, 2022-12-07:
OK #3330 merged for v1.9.0 release - will run ip2 by default for bitcoind to have a fallback sync network when Tor is jammed.

closed 2022-12-15 by rootzoll