raspiblitz/raspiblitz #2420: Make paths easy to change

nyxnor, 2021-07-15:
Paths are hard coded, if using bare metal, just like a .conf, a .lib to define all paths would work
https://github.com/nyxnor/raspiblitz/blob/e8b1be2ccbcce9a264771f1b8887326a8517633e/home.admin/config.scripts/tor.functions.lib#L61
Sourcing just like the raspiblitz.conf, we would need to change ahrd paths to variables.

Currently, problems appears when trying to use NVME or CM4, but this could be used to adapt to any platform.

nyxnor, 2021-07-22:
just like config.scripts/tor.functions.lib from PR 2419

rootzoll, 2021-08-16:
I would try to prevent sourcing pathes from outside files as much as possible and let the script itself detect the evironment its runnig on (or get that info from the raspiblitz.info) and then set the path dynamicly in the script.