raspiblitz/raspiblitz #2423: /home/admin/00infoblitz.sh: line 383: cannot create temp file for here-document: No space left on device

ronledfordjr, 2021-07-16:
Hello, I am receiving this message on all four of my nodes.

[image]

https://termbin.com/qxdg

openoms, 2021-07-17:
```
/dev/root      ext4       15G   14G     0 100% /
```
Your root partition (on the SDcard) has filled up.

It is recommended to use a 32 GB card.

You can find out what is taking up space with the linux command:
```
sudo du -hd1 /
```
explore paths going deeped in the directory tree.

Flashing the v1.7.0 image new on your existing card or a new 32Gb card would solve this.

ronledfordjr, 2021-07-17:
ordering 32GB sd cards. I did reboot all nodes and message went away. Thanks

closed 2021-11-01 by ronledfordjr