raspiblitz/raspiblitz #2424: blitz.local

rootzoll, 2021-07-17:
To make it easier to call the RaspiBlitz from the local web browser - it should advertise itself as e.g. `blitz.local`

See deatils on this in this article:
https://www.howtogeek.com/167190/how-and-why-to-assign-the-.local-domain-to-your-raspberry-pi/

openoms, 2021-07-18:
This was a privacy concern discussed in: https://github.com/rootzoll/raspiblitz/issues/819

Need to assume that ISP provided routers are leaking the device names to the ISP.

Currently the base image name is untouched so it would be `raspberrypi.local` or other default if a custom image was used.

rootzoll, 2021-07-18:
> Need to assume that ISP provided routers are leaking the device names to the ISP.

Yeah I always have that blind spot because I am free to chose my own router - not everybody is priviledged to have.

openoms, 2022-01-16:
This could be closed depending on the outcome of #2890

rootzoll, 2022-01-16:
Yepp that also came to my mind ... technically for a `blitz.local` the raspiblitz announces itself on the network. I mean we could go without "blitz.local" by default - because by default the Blitz has a LCD display to tell the user the local IP.

rootzoll, 2022-03-09:
OK .. deciding to not go with the `blitz.local` to preserve privacy in the local network and possible ISP router. The Raspiblitz has the LCD Display  by default - that gives the IP to enter into browser. - that should work for now. Closing issue.

closed 2022-05-09 by rootzoll