raspiblitz/raspiblitz #2433: raspiblitz.conf permissions

rek79, 2021-07-21:
the raspiblitz.conf file is created under /mnt/hdd with permissions root  root -wr-r--r--r.  When installing bonus features such as sphinxrelay there are permission denied messages since the script is running under admin.   Line 158 of bonus.sphinxrelay.sh does not use sudo to make edits.  It seems like raspiblitz.conf should be owned by admin:admin.  Thoughts?

[image]

[image]

rek79, 2021-07-21:
If this is acknowledged as an issue then this may be a potential solution using "tee" command:

Source:
https://www.cyberciti.biz/faq/sudo-append-data-text-to-file-on-linux-unix-macos/

Method 1: Use tee command
The tee command read from standard input (such as keyboard) and write to standard output (such as screen) and files. The syntax is:
```
echo 'text' | sudo tee -a /path/to/file
echo '192.168.1.254 router' | sudo tee -a /etc/hosts
```

Sample outputs:

Password:
`192.168.1.254   router`
This solution is simple and you avoided running bash/sh shell with root privileges. Only append or write part needed root permission.

rek79, 2021-07-21:
Confirmed working fix:
[image]
[image]
`echo "BTCRPCexplorer=off" | sudo tee -a /mnt/hdd/raspiblitz.conf`

Should I prepare a PR to replace "`>>`" operators with " `| sudo tee -a`"?

openoms, 2021-07-21:
Are you now building with the branch `pre-1.7.1` ? Because this clearly was not a problem in v1.7.0.

rek79, 2021-07-21:
No.  I'm still using 1.7.  It could easily go unnoticed in v1.7 since the script is just establishing defaults in the config file.  I believe it would not stop the bonus features from actually functioning.  If you are running 1.7 and try typing `echo "BTCRCexplorer=off" >> /mnt/hdd/raspiblitz.conf` as admin you should get the same result.  This command would require logging in as root to be successful.

openoms, 2021-07-21:
on my vanilla v1.7 node:
```
admin@David:~ $ ls -la /mnt/hdd/raspiblitz.conf 
-rwxrwxrwx 1 root root 367 Jul 21 22:01 /mnt/hdd/raspiblitz.conf
```

the `/mnt/hdd/raspiblitz.conf` is owned by root, but writable by all users.
The problem you encounter is that on your system the `raspiblitz.conf` has stricter permissions.

rek79, 2021-07-22:
I should have noticed that. The config file must have been created by a different branch of code when I had trouble with localIP issue on Odroid HC2.  I'll review the code to see where this could have happened and report back.  We may need to check permissions after config creation if it doesn't happen normally at 40addHDD.sh

rek79, 2021-07-22:
I looked through the code.  Every location that creates raspiblitz.conf seems to set 777 permissions right after.  I manually changed the permissions using chmod 777.  I'm not sure how they got altered.  I will most likely save blockchain data, wipe and start with a fresh image.  The most likely culprit is user error on my part while tinkering.  I'm going to close this issue.

openoms, 2021-07-22:
when a script is run with `sudo` it runs with root user
so a command like:
`echo "anything" >> raspiblitz.conf`
would create the file with default permissions (not 777) if it wouldn't exit before.
It was probably a wrong order of scripts with some failing.

rek79, 2021-07-22:
That makes a lot of sense.  I did manually run individual scripts with sudo to troubleshoot.  It was a valuable lesson learned, thanks.

rootzoll, 2021-08-24:
The bootstrap script now makes sure that the `raspiblitz.conf` gets the chmod 664 with owner=root & group=sudo (admin user is in group sudo) this way every config-script called by root or admin can simply edit/write to the `raspiblitz.conf` but other users like additional apps can just read and not write to it. I think thats a good balance of security for now.

rootzoll, 2021-08-24:
This will be part of the v1.7.1rc1. To be tested if:
- [ ] after fresh setup raspiblitz.conf if 664 with root:sudo
- [ ] after recover/update raspiblitz.conf if 664 with root:sudo
- [ ] after migration raspiblitz.conf if 664 with root:sudo
- [ ] other scripts running fine with the new permission

openoms, 2021-09-27:
@rootzoll in af3db7f I added the `cl.check.sh` which makes sure there are no config entries in the CL config which would stop it running. 
It runs with the `bitcoin` user which has no sudo access, but can uninstall the problematic plugins.

Can we add 777 permissions on raspiblit.conf or can you think of a better solution to be able to swithc those plugins off in the config if they are deactivated by `cl.check.sh`?

rootzoll, 2021-09-28:
I would like to keep the `raspiblitz.conf` not writable to other users - especially so that other additional apps can just read it, but not write it. We can think about the `bitcoin` user to  the `sudo` group ... but the whole idea of having a seperate `bitcoin` user was to keep it seperated from the `admin` user with sudo rights - but that would be at least better than just setting `777` permissions on the `raspiblitz.conf` file.

rootzoll, 2021-09-28:
Maybe having a seperate file for the cl plugins that can then be part of the CL data dir makes more sense anyway for the CL plugins? That way it would keep a clean seperation.