raspiblitz/raspiblitz #2434: LNDhub implementation (200k sat bounty)

godSaysHODL, 2021-07-22:
Really like the LNDhub implementations on myNode and Umbrel. Having an open source bank is dope and brings lots of scaling benefits to family uncle-Jim setups, with the clean look of Bluewallet's interface.

If you can get this project merged into RaspiBlitz, i'll pay a 200k sat bounty 🙌. 

After this is complete i'll be migrating to the Raspiblitz platform as my main node of choice 🤝

nyxnor, 2021-07-22:
If someone wants to work on this, there is this unfinished code I made to implement lndhub without docker https://github.com/rootzoll/raspiblitz/issues/230#issuecomment-847757874

openoms, 2021-07-23:
Try the existing LNbits extension, it might work already.

xanoni, 2021-08-02:
This might 

> If someone wants to work on this, there is this unfinished code I made to implement lndhub without docker [#230 (comment)](https://github.com/rootzoll/raspiblitz/issues/230#issuecomment-847757874)

This might fix it: https://github.com/rootzoll/raspiblitz/issues/230#issuecomment-890838807

openoms, 2021-08-31:
@godSaysHODL did you try to install LNBits in which there is a functional LNDHub extension already?

Can also be tested on https://lnbits.com (this is a custodial demo wallet on Opennode). The biggest difference is that your own node would not be available on a public domain, but I think BW and Zeus can connect over Tor already.

@xanoni you could try this too.

godSaysHODL, 2021-09-01:
@xanoni 

Definitely still gonna pay the bounty. Just ping me here when it's reviewed and looks like it'll be merged and I'll donate to whoever you choose!

godSaysHODL, 2021-09-01:
> @godSaysHODL did you try to install LNBits in which there is a functional LNDHub extension already?
>
> Can also be tested on https://lnbits.com (this is a custodial demo wallet on Opennode). The [...]

I will check this out also. 👌

krtk6160, 2021-11-01:
> Yeah I saw your comment already a few weeks ago, but I had already set up LndHub, so I never gave it a try. The LndHub script is almost done so I'll get that across the finish line. If after all this time there's still someone willing to pay a bounty, please donate it to RaspiBlitz or to one of the upstream projects ;-)

Do you still plan on finishing this? LNDHub support would be awesome :)

xanoni, 2021-11-01:
Oh man, I keep remembering this ticket all the time and feel a lot of guilt and shame (really!) ... thank you for pinging me. I'm not making promises because I already did that too often here and failed to deliver afterwards ;-)

There's no specific problem, there's just an infinite amount of other stuff. I just haven't prioritized time to continue.

prusnak, 2022-07-13:
There is a very promising reimplementation of LndHub in Go language here: https://github.com/getAlby/lndhub.go
I recommend to use this as a LndHub/BlueWallet server.