raspiblitz/raspiblitz #2435: Unable to re-install loop

vmalarcon, 2021-07-22:
I'm loving my RaspiBlitz but I'm an idiot! :-(

I installed Loop and I was able to use it for a while until I decided to uninstall it. Now, if I try to re-install it I see the below (I had to freeze the terminal because it goes by very fast)

```
dialogcancel(0)
RTL Webinterface Setting unchanged.
BTC-RPC-Explorer Setting unchanged.
Cryptoadvance Specter Setting unchanged.
ElectRS Setting unchanged.
BTCPayServer setting not changed.
lndmanage setting unchanged.
chantools setting unchanged.
Balance of Satoshis setting unchanged.
PyBLOCK setting unchanged.
ThunderHub setting unchanged.
LNbits setting unchanged.
LIT Setting changed ..
making sure the lit service is not running
# INSTALL LIGHTNING TERMINAL
adduser: The group `lit' already exists.
Failed to start lnbits.service: Unit lnbits.service not found.
Can't open /home/lit/.lit/tls.cert for reading, No such file or directory
548220057472:error:02001002:system library:fopen:No such file or directory:../crypto/bio/bss_file.c:69:fopen('/home/lit/.lit/tls.cert','r')
548220057472:error:2006D080:BIO routines:BIO_new_file:no such file:../crypto/bio/bss_file.c:76:
unable to load certificate
please wait ...
Sphinx Relay unchanged.
```

Now I checked and that folder (/home/lit) is no longer there so it makes sense that there are no files. How can I fix this?

The steps to reproduce would be:
1) Install Loop
2) De-install Loop
3) Try to re Install Loop

Is there some script I can run to 'force install' Loop again? Thanks!

nyxnor, 2021-07-22:
```
sudo userdel lit
bash config.scripts/bonus.lit.sh on
```

vmalarcon, 2021-07-23:
Thank so much! I had to add a delete group to the fix as well... so:
```
sudo userdel lit
sudo groupdel lit
bash config.scripts/bonus.lit.sh on
```

closed 2021-07-23 by vmalarcon