raspiblitz/raspiblitz #2437: Idea — add BlockSci Chain Analytics (Python interface)

xanoni, 2021-07-23:
This one might be a bit controversial, but I thought it would be fun to have BlockSci (https://github.com/citp/BlockSci) on the Pi. 

It's an open source chain analytics solution in the form of Python libraries. I think it could be useful for learning purposes and to better understand one's own onchain footprint.

I've started deploying it to my Pi a few days ago and want to start playing with it soon... compilation worked fine with some minor tweaks due to dependency changes... will report back here, but feel free to let me know what you think and if you're aware of any similar projects.

TODO:
— Run the binaries on the blockchain to see if it actually works :)
— Benchmark performance against a fully powered 8+ core computer to see if it's worth it
— Test how useful the RPC mode is, where you run the Python on a remote computer and just use the Pi as a backend (they say it's slower than accessing disk directly, but unclear how slow)

nyxnor, 2021-07-23:
No problem about the tool, but the development stopped:
_As of November 2020, we are no longer actively developing or supporting BlockSci. Please be aware that as cryptocurrencies continue to evolve, they may lose compatibility with BlockSci and might break it in unexpected ways._

xanoni, 2021-07-23:
Yes I saw that and should have mentioned it ... you are right that it will break at some point, and it could be as soon as taproot. (I have no indication that it will, just guessing.)

It's still worth it to me. Maybe it can be fixed easily once it breaks.

nyxnor, 2021-07-23:
No deprecated software will be implemented in the mainstream Blitz repo, it bringes insecuritites, unpatched code, new bugs found that were not fixed, exploits. You are free to work on a install script for your forked repo.
Another alternative is to find a well maintained fork of that project.

xanoni, 2021-07-23:
No problem, I don't really care at all about whether it's in the main repo. As long as people who are curious can easily access it.

xanoni, 2021-08-23:
Given this can not become an official component of this repo, I'll close the ticket for now.

closed 2021-08-23 by xanoni