raspiblitz/raspiblitz #2438: Feature request: Allow user to use TOR only for outgoing connections

jchia, 2021-07-23:
Some users may prefer regular Internet over TOR because of concerns about latency and reliability and prefer to use regular Internet. But they still want to connect to onion-only peers, so they still need TOR for outgoing connections to them.

Can we enable a mode where TOR is enabled but used only to connect to onion-only peers? In this case we will also publish only the Internet address of the lightning node but not the onion address.

nyxnor, 2021-07-23:
> Some users may prefer regular Internet over TOR because of concerns about latency and reliability and prefer to use regular Internet. But they still want to connect to onion-only peers

This wouldnt help speed, in fact, reaching a hidden service could be the same speed as reaching a plain net domain if one party is using Tor.

> Can we enable a mode where TOR is enabled but used only to connect to onion-only peers?

change in bitcoin.conf to `onlynet=onion`
https://github.com/bitcoin/bitcoin/blob/master/doc/tor.md#1-run-bitcoin-core-behind-a-tor-proxy

>  In this case we will also publish only the Internet address of the lightning node but not the onion address.

I have no idea what you said. Rephrase this pls.

xanoni, 2021-07-23:
> I have no idea what you said. Rephrase this pls.

I think he is saying that the own node would no longer be running a HS, so it wouldn't make sense to publish an .onion address.

> This wouldnt help speed, in fact, reaching a hidden service could be the same speed as reaching a plain net domain if one party is using Tor.

Well, there are a few scenarios where it would help ... most prominently if the other node is also a non-Tor node.

What happens if a Tor node connects to our non-Tor node? Assume this would also help, as there would only be 3 hops instead of 6 hops, correct?

nyxnor, 2021-07-23:
> most prominently if the other node is also a non-Tor node.

But he said he wants only to connect to tor nodes, hence onlynet=onion

> What happens if a Tor node connects to our non-Tor node? Assume this would also help, as there would only be 3 hops instead of 6 hops, correct? Also the Rendezvous procedure might be shorter.

Less hops, yes, less relays, but, it as onion services are native to Tor, they are resolved faster. Reaching plain net through Tor brings more load to the relays.

xanoni, 2021-07-23:
> But he said he wants only to connect to tor nodes, hence onlynet=onion

I think this was phrased a bit ambiguously .. I think it was supposed to say "use Tor for Tor peers, use IPv4/IPv6 for all other peers".

> Less hops, yes, less relays, but, it as onion services are native to Tor, they are resolved faster. Reaching plain net through Tor brings more load to the relays.

I haven't benchmarked this, but overall I agree that NOT using Tor is probably NOT worth it, given the privacy-by-default benefit.

jchia, 2021-07-24:
> wants only to connect to tor nodes

Despite preferring regular Internet, I still want to connect to onion-only nodes, i.e. nodes that have only an onion address but no Internet address. Currently onion is a binary choice in Raspiblitz.  "want to connect to only onion nodes" vs " want to connect to onion-only nodes".

The goal is simply to use regular Internet instead of Tor wherever possible and use Tor only if regular Internet is not an option, i.e. to connect to nodes that have only an onion address. For incoming connections, to maximize regular Internet over Tor, I also do not want to publish an onion address that can be used to connect to my node.

xanoni, 2021-07-25:
Are there any routing performance issues with Tor? Maybe I'm just not aware of the trade-offs here.

jchia, 2021-07-25:
> Are there any routing performance issues with Tor? Maybe I'm just not aware of the trade-offs here.

Tor being an extra layer over regular Internet is slower and less reliable than regular Internet. Are there mitigating factors to help routing over Tor be as good as routing over regular Internet?

xanoni, 2021-07-25:
Of course it is slower. But are the extra 250 milliseconds a problem for the Lightning use case?

I haven't read anything to that extent. But if if missed it, would appreciate a link.

rootzoll, 2021-08-17:
> The goal is simply to use regular Internet instead of Tor wherever possible and use Tor only if regular Internet is not an option

The decided some time ago to go Tor by default with RaspiBlitz - as much as possible and its a good starting default: You can always reduce your privcacy later on - but not the other way around once you e.g. let a footprintof your IP together with your nodeid somewhere. 

There is still the option to deactivate Tor if you really prefer. Basically there it could also be a "hybrid" mode - but thats not high on our prio right now to have that as a standard feature. @jchia You can experiment with editing your LND config manually and report your experience if you like. 

I will leave this issue open for now if any other users like to express that this is an important feature to them too.

jchia, 2021-08-21:
For now, I'm advertising both an IP address and onion address by adding an `externalip` to lnd.conf after enabling `runBehindTor` in raspiblitz.conf. I think in this case, if I connect to an IP address I will still go through TOR, but I can still accept incoming Internet connections.

xanoni, 2021-08-22:
You would also have to enable listening on IP ... for IPv4:
```
listen=0.0.0.0:9735
```

instead of

```
listen=localhost
```

(unless RaspiBlitz already does that automatically)

jchia, 2021-08-22:
@xanoni It seems to be not enough. lnd is still not listening on 0.0.0.0:9735:
```
admin@raspberrypi:/mnt/hdd $ pgrep -fla bin/lnd
16050 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --tor.socks=9070 --tor.control=9071 --listen=127.0.0.1:9735 --accept-keysend
$ sudo netstat -tlnp | grep ':9735.*16050/lnd'
tcp        0      0 127.0.0.1:9735          0.0.0.0:*               LISTEN      16050/lnd
admin@raspberrypi:/mnt/hdd $ grep ^listen lnd/lnd.conf
listen=0.0.0.0:9735
```

xanoni, 2021-08-22:
It's probably this script:
https://github.com/rootzoll/raspiblitz/blob/9c7923dc508e39ebd72b7b914d573eaa16b864ed/home.admin/config.scripts/internet.tor.sh#L174

It overwrites the `lnd` launch parameters in `/etc/systemd/system/lnd.service`. I'm not sure what the RaspiBlitz-way to fix this is, but you could just edit that file directly ...