raspiblitz/raspiblitz #2439: Trying to use a screen that works via spi, throws and error

llvllch, 2021-07-23:
Hey there! I've ran into a bit of a wrinkle and I'm hoping that the hive-mind might be able to help.

In setting up for the screen to be used, I've enabled spi via raspi-config, but when I try to run my code, I'm getting:

FileNotFoundError: [Errno 2] No such file or directory: '/dev/spidev0.0'

Any words of wisdom before I throw myself down the rabbit hole?

nyxnor, 2021-07-23:
https://raspberrypi.stackexchange.com/questions/69911/how-to-make-raspbian-load-the-spi-module-on-boot-up

llvllch, 2021-07-23:
Thanks!

It's there, but still not working. The internet has inconsistent info, but I am not sure that the spidev module is installed in the kernel in this image. Is that possible?

nyxnor, 2021-07-23:
It needs to be uncommented, check inside
`sudo nano /boot/config.txt`if that line is uncommented (withtout # at the beggining, then reboot). If it is not this, I don't know.

llvllch, 2021-07-23:
> It needs to be uncommented, check inside
> `sudo nano /boot/config.txt`if that line is uncommented (withtout # at the beggining, then reboot). If it is not this, I don't know.

Yup, it's commented. I can't see the module even listed so I have a feeling it might be missing. Thanks though..... I'll dig a little more

llvllch, 2021-07-23:
Solved! 
```
 sudo apt install python-spidev python3-spidev
```
followed by adding admin to the spi group did the trick.
```
 sudo usermod -a -G spi admin
```
All has been running fine for a couple of hours, so I don't think I managed to break anything.

closed 2021-07-23 by llvllch