raspiblitz/raspiblitz #2444: Installing LiT fails: !!! LND PGP Verify not OK

The34Goon, 2021-07-26:
```
admin@Goon34:~ $ bash config.scripts/bonus.lit.sh on
making sure the lit service is not running
# INSTALL LIGHTNING TERMINAL
Lege Benutzer »lit« an ...
Lege neue Gruppe »lit« (1007) an ...
Lege neuen Benutzer »lit« (1011) mit Gruppe »lit« an ...
Das Home-Verzeichnis »/home/lit« existiert bereits. Kopiere keine Dateien aus »/etc/skel«.
###### SYNCING MACAROONS, RPC Password AND TLS Certificate ######
# make sure LND app-data directories exist

# copy macaroons to central app-data directory and ensure unix ownerships and permissions
# make sure admin has a symlink at ~/.lnd to /mnt/hdd/app-data/lnd/
# make sure LND conf is readable and symlinked
# make sure TLS certificate is readable and symlinked
# fix the macaroon for LNbits
make sure symlink to central app-data directory exists
# OK - macaroons written to /home/lnbits/lnbits/.env
# persist settings in app-data
# make sure the data directory exists
# symlink
# move the standalone Loop and Pool data to LiT
# Loop
# remove so can't be used parallel with LiT
making sure the loopd.service is not running
# Loop is not installed.
# make sure the data directory exists
# symlink
# Pool
# remove so can't be used parallel with LiT
# making sure the service is not running
# Pool is not installed.
# make sure the data directory exists
# symlink
Detect CPU architecture ...
OK running on aarch64 architecture.
--2021-07-26 02:27:35--  https://github.com/lightninglabs/lightning-terminal/releases/download/v0.4.1-alpha/manifest-v0.4.1-alpha.txt
Auflösen des Hostnamens github.com (github.com)… 140.82.121.3
Verbindungsaufbau zu github.com (github.com)|140.82.121.3|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://github-releases.githubusercontent.com/252564593/42298a80-7b9e-11eb-82a4-685502fabc15?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210726%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210726T002735Z&X-Amz-Expires=300&X-Amz-Signature=7c7c7c4b223496a390ecd71d8a7f5c0fd4f8e7186e33b6952c3a6828145fe014&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=252564593&response-content-disposition=attachment%3B%20filename%3Dmanifest-v0.4.1-alpha.txt&response-content-type=application%2Foctet-stream [folgend]
--2021-07-26 02:27:36--  https://github-releases.githubusercontent.com/252564593/42298a80-7b9e-11eb-82a4-685502fabc15?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210726%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210726T002735Z&X-Amz-Expires=300&X-Amz-Signature=7c7c7c4b223496a390ecd71d8a7f5c0fd4f8e7186e33b6952c3a6828145fe014&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=252564593&response-content-disposition=attachment%3B%20filename%3Dmanifest-v0.4.1-alpha.txt&response-content-type=application%2Foctet-stream
Auflösen des Hostnamens github-releases.githubusercontent.com (github-releases.githubusercontent.com)… 185.199.110.154, 185.199.111.154, 185.199.109.154, ...
Verbindungsaufbau zu github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.110.154|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 1237 (1.2K) [application/octet-stream]
Wird in »manifest-v0.4.1-alpha.txt« gespeichert.

manifest-v0.4.1-alpha.txt                        100%[==========================================================================================================>]   1.21K  --.-KB/s    in 0s

2021-07-26 02:27:36 (15.1 MB/s) - »manifest-v0.4.1-alpha.txt« gespeichert [1237/1237]

# LiT v0.4.1-alpha for arm64
# SHA256 hash: 0e73605ca44dfa0c8b8406776562780a2687c531351145e24b302e0240d5f496

# get LiT binary
--2021-07-26 02:27:36--  https://github.com/lightninglabs/lightning-terminal/releases/download/v0.4.1-alpha/lightning-terminal-linux-arm64-v0.4.1-alpha.tar.gz
Auflösen des Hostnamens github.com (github.com)… 140.82.121.3
Verbindungsaufbau zu github.com (github.com)|140.82.121.3|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://github-releases.githubusercontent.com/252564593/3b027c80-7b9e-11eb-880c-269189689e50?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210726%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210726T002736Z&X-Amz-Expires=300&X-Amz-Signature=cabca12a15519742ecec2e3ba84a357f48f0f45dbe61aeb70adc56113d074787&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=252564593&response-content-disposition=attachment%3B%20filename%3Dlightning-terminal-linux-arm64-v0.4.1-alpha.tar.gz&response-content-type=application%2Foctet-stream [folgend]
--2021-07-26 02:27:36--  https://github-releases.githubusercontent.com/252564593/3b027c80-7b9e-11eb-880c-269189689e50?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210726%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210726T002736Z&X-Amz-Expires=300&X-Amz-Signature=cabca12a15519742ecec2e3ba84a357f48f0f45dbe61aeb70adc56113d074787&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=252564593&response-content-disposition=attachment%3B%20filename%3Dlightning-terminal-linux-arm64-v0.4.1-alpha.tar.gz&response-content-type=application%2Foctet-stream
Auflösen des Hostnamens github-releases.githubusercontent.com (github-releases.githubusercontent.com)… 185.199.110.154, 185.199.111.154, 185.199.109.154, ...
Verbindungsaufbau zu github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.110.154|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 41604947 (40M) [application/octet-stream]
Wird in »lightning-terminal-linux-arm64-v0.4.1-alpha.tar.gz« gespeichert.

lightning-terminal-linux-arm64-v0.4.1-alpha.tar. 100%[==========================================================================================================>]  39.68M  13.1MB/s    in 3.0s

[... 30 lines cut ...]
# check gpg finger print
gpg: WARNING:  Kein Kommando angegeben.  Versuche zu raten was gemeint ist ...
pub   rsa4096/8E4256593F177720 2018-05-02 [SC] [verfällt: 2034-04-28]
      F4FC70F07310028424EFC20A8E4256593F177720
uid                           Oliver Gugger <gugger@gmail.com>
sub   rsa4096/5E50C90D766DB9FC 2018-05-02 [E] [verfällt: 2034-04-28]
sub   rsa4096/6376F16B3C9F08CC 2019-09-25 [E] [verfällt: 2021-09-24]
sub   rsa2048/03DB6322267C373B 2019-09-25 [S] [verfällt: 2021-09-24]
sub   rsa2048/4F50789ED785C6D4 2019-09-25 [A] [verfällt: 2021-09-24]
sub   rsa4096/5F227E08FA339C20 2021-04-24 [] [verfällt: 2023-04-24]
sub   rsa4096/63B65BEF7899D597 2021-04-24 [] [verfällt: 2023-04-24]
sub   rsa4096/48F5EB2E7E59AF8C 2021-04-24 [] [verfällt: 2023-04-24]
sub   rsa4096/F0A0ED07755FDBCF 2021-05-12 [E] [verfällt: 2023-05-12]
sub   rsa4096/988BB7626335E3FB 2021-05-12 [S] [verfällt: 2023-05-12]
sub   rsa4096/7BBD8D4D95DB9F03 2021-05-12 [A] [verfällt: 2023-05-12]
gpg: key 8E4256593F177720: 1 Beglaubigung wegen fehlendem Schlüssel nicht geprüft
gpg: Schlüssel 8E4256593F177720: "Oliver Gugger <gugger@gmail.com>" nicht geändert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:                             unverändert: 1
goodSignature(0)
correctKey(1)

!!! BUILD FAILED --> LND PGP Verify not OK / signature(0) verify(1)
```
[https://termbin.com/dmoj](url)

rootzoll, 2021-08-17:
Adding this to recheck before v1.7.1 release

rootzoll, 2021-08-17:
v1.7.1rc1 will already contain this LIT & pool update #2331

closed 2021-08-18 by The34Goon