raspiblitz/raspiblitz #2446: 502 Nginx Error When Deploying Testnet BTCPay

lionkingaw, 2021-07-27:
I have installed a fresh testnet BTCPay server on Lunanode and receive a 502 error.

I then installed a fresh mainnet BTCPay server on Lunanode, and then switched to testnet using the commandline, and again I received a a 502 error.

I installed a regular mainnet BTCPay server and this functions properly.

There is a bug when trying to deploy a testnet server, or changing a miannet server to testnet.

Thanks

rootzoll, 2021-08-17:
So you are running a RaspiBlitz on a Luna Cloud KVM Instance?

Will put issue to v1.7.1 to retest BTCPay when running testnet.

rootzoll, 2021-09-21:
With v1.7.1 we generally changed how testnet works on RaspiBlitz ... it now runs in parallel to mainnet. Just running Testnet might comeback in the future. Closing issue for now.

closed 2021-09-21 by rootzoll