raspiblitz/raspiblitz #2447: Lightning Pool Bug: Unable to fetch latest batch

rek79, 2021-07-27:
I have Lightning Pool up and running.  The account was successfully created.  I was also able to create an ask order.  However, the order is never accepted even though the premium is within market range.  This are the errors periodically received:
[image]

rek79, 2021-07-27:
Are those errors normal and possibly the main reason order is not accepted is due to node tier T0 and offered liquidity only 400,000 sats?

openoms, 2021-07-28:
Try restarting the service.
`sudo systemctl resatrt litd`

you should see the orders regardless of your node status or funds.

rek79, 2021-07-28:
I tried restarting litd.  That caused litd service to startup and then fail.  I rebooted which seemed to speed up load times on Thunderhub and refresh on Lightning Pool.  I opening an issue at https://github.com/lightninglabs/pool/issues/274#issue-954287177.  One comment is leading to Tor connection being to slow to respond.  I'm wondering if I can change some settings in config while still keeping TOR only active.

rek79, 2021-08-05:
Update: I was able to fill a bid order.   I'm still not able to fill an ask order but, from what I hear, this is due to low demand and T1 nodes taking precedence.  I occasionally receive the "unable to fetch batch" warnings but that is probably due to slow Tor connection.  I'll close this issue out.

closed 2021-08-05 by rek79