raspiblitz/raspiblitz #2450: Inverted colors on Waveshare 3,5" B rev 2.0 LCD

stepansonsky, 2021-07-29:
Hello, I've bought Waveshare 3,5" B rev 2.0 LCD and unfortunately colors were inverted. So I've found [this issue](https://github.com/rootzoll/raspiblitz/issues/493) with solution, which broke my RaspiBlitz. When booting, colors were ok and I can see running boot script fine, until RaspiBlitz splash screen shown. I was able to connect through SSH and normally browse the menu and control the device, but RaspiBlitz logo stuck on the screen forever.

After few hours of reinstalls and hacking, I've found definitive and easy solution:

- Download [waveshare35b-v2-overlay.dtb](https://github.com/waveshare/LCD-show/blob/master/waveshare35b-v2-overlay.dtb)
- Rename it to **waveshare35b-v2.dtbo**
- Copy it into **overlays** dir on SD card
- In **config.txt** replace `dtoverlay=waveshare35a:rotate=90` by `dtoverlay=waveshare35b-v2:rotate=90`

Could you please add this fix into the next release?
Thanks for your great work!

Dehumanizer77, 2021-11-25:
Thanks a lot! I had the same problem, now fixed :-)

rootzoll, 2021-11-25:
@stepansonsky thanks for finding out how to get the alternative display version running. But to support it by default we have the problem that as long we cannot detect the display type attached its hard to install the correct drivers.

Does anybody has an idea how to detect the display type attached?

Until we have some kind of automatic detection alternative displays can only be set manually. For this there could multiple ways which I am also not sure what will the best to choose from:

- the display type could be set during sd card build - but that would mean that people with alternative display need to build their own sd card images - might not be the most practical solution.
- the display type could be set by placing a special file to the boot section of the sd card from your laptop before starting the sd card on the pi - still very technical
- the display could be choosen during setup process - this might be a bit confusing to new users, so this might only be part of the SSH setup in the future and not of the upcomming WebUI browser setup

At least once the alternative display was set it can then be stored as a setting in the `raspiblitz.conf` so that on future sd card updates/recoveries the matching driver will get installed automatically ... but this then still needs to be implemented in certain scripts ... to be mentioned here: `/alternative.platforms/display.alternatives.sh` & `/config.scripts/blitz.display.sh`

I will mark this issue for v1.8.1 .. its still a bit up the road but around that version adding an ePaper display alternative was already an idea. So it would make sense to look at this at this point in time again and make a solution that works for a set of alternative displays. Until then running RaspiBlitz with alternative displays types will stay a bit hacky and manually repeated on every update.

stepansonsky, 2021-12-01:
Thank you for paying attention to the issue. I think the best would be some picker in settings. Or MVP solution - just add the driver into the image and add instructions to the readme.md, how to fix inverted colours on the Waveshare 3,5" B rev 2.0 by editing the config.

RyanWoirol, 2022-04-13:
This worked perfect.  Thank you so much

flyerbear, 2023-03-12:
I have the same issue and this solution fixes it every update. I'd like to incorporate these steps into my `custom-installs.sh` file, but I was hoping someone could double-check my bash script, especially the find/replace of the text in `config.txt`. The steps required for this to work would be:
1. Rename `waveshare35b-v2.dtbo` and copy into `overlays` as described by @stepansonsky.
2. Copy `waveshare35b-v2.dtbo` into `/mnt/hdd/app-data/custom/`.
3. Add the following lines to `/mnt/hdd/app-data/custom-installs.sh`:

```
cp /mnt/hdd/app-data/custom/waveshare35b-v2.dtbo /boot/overlays/
sed -i -e 's/waveshare35a/waveshare35b-v2/g' /boot/config.txt
```

Assuming the above bash script lines are correct, all anyone affected by this would have to do is rename and copy the file as @stepansonsky directed, edit `config.txt` once, then additionally copy the .dtbo file into `/mnt/hdd/app-data/custom/`, edit `custom-installs.sh` as above, and the change should persist through updates.

flyerbear, 2023-03-13:
For what it's worth, I included the above lines in `custom-installs.sh` and upgraded using the 1.9.0rc2 SD card 'minimal' image. The screen's colors were correct, but it was upside down and the height seems wrong - text is "squished" vertically. I was able to flip it back around by manually setting `lcdrotate=90` in `config.txt`, but I can't figure out how to get the resolution correct. Is there a way to get back the touchscreen and display options in the menu if you do a minimal install?

Also, `config.txt` seems different in 1.9 than it did in 1.8. The dtoverlay line appears three times instead of just once. Should the modification be handled differently in 1.9?

flyerbear, 2023-03-14:
Ok, I think I figured it out, and why config.txt seemed different. Need to first run `lcd` before applying the changes, but there is a y/n prompt to respond to when you do. I'm not sure, but I think the So the custom install script should be:

```
echo "y" | lcd
cp /mnt/hdd/app-data/custom/waveshare35b-v2.dtbo /boot/overlays/
sed -i -e 's/waveshare35a/waveshare35b-v2/g' /boot/config.txt
```
Is that right?

Dehumanizer77, 2023-05-25:
This fix has stopped working for me in 1.9.0
When I replace waveshare35a by waveshare35b-v2 the display stays dark even after fully booted.

Dehumanizer77, 2023-05-26:
Oh, I just incorrectly downloaded the file (it had HTML inside :-)
So, the fix still works, sorry for the confusion.