raspiblitz/raspiblitz #2452: Closing channel is not proceeding

clockshell, 2021-07-31:
Hello!

My node was offline for a while, so ACINQ closed the channel to my node. It is yet still pending to close. ACINQ claims that my node has not picked up the closing confirmation.

So now I am using Raspiblitz 1.7 on this node:

`03eea3e2115e1d2c1038e0826f36dabc1b91d019445c07ecfc3a3b0930f285620e@au4y64saizeewrqp6cizbpegtf5tsgif74p6vi4y72ux2fyl4sczsyqd.onion:9735`

This is the output of lncli pendingchannels:

`"waiting_close_channels": [ { "channel": { "remote_node_pub": "03864ef025fde8fb587d989186ce6a4a186895ee44a926bfc370e2c366597a3f8f", "channel_point": "aa96601ce517e070563b0aabe964d228fc91098e1d0f0f031d1bce39c30ef443:1", "capacity": "403645", "local_balance": "0", "remote_balance": "0", "local_chan_reserve_sat": "0", "remote_chan_reserve_sat": "0", "initiator": "INITIATOR_LOCAL", "commitment_type": "LEGACY" }, "limbo_balance": "0", "commitments": { "local_txid": "", "remote_txid": "", "remote_pending_txid": "", "local_commit_fee_sat": "0", "remote_commit_fee_sat": "0", "remote_pending_commit_fee_sat": "0" } } ]`

How can I get my node to pick up the closing confirmation? It seems strange to me that any of the information seem to have gotten dumped and the only ones left are remote_node_pub, channel_point and capacity.

Best regards,

clockshell

rootzoll, 2021-08-17:
@openoms can you take a look?

openoms, 2021-08-17:
can see the channel on https://ln.fiatjaf.com/node/03eea3e2115e1d2c1038e0826f36dabc1b91d019445c07ecfc3a3b0930f285620e as force-closed.

I presume 372168 sats should have hit your wallet a while ago https://blockstream.info/address/bc1q9j9zpxtzywv8ly98s7s5s853nqtuxm4ym40pex

Check:
`lncli listchaintxns`

Balanc of satoshis:
`bos closed`
is able to give more detailed info.

clockshell, 2021-08-22:
lncli listchaintxns outputs pretty much, so I put it into a pastebin: https://pastebin.com/EiuBNCqD

This is the output of bos closed of user bos. There is no ACINQ listed:
`$ bos closed 

closes: 
  - 
    peer_public_key:       02360f8e4f4181892136af6fe302c4017fb1e57672505e8584ae2acac87792deb6
    peer_alias:            fishfeed
    is_remote_force_close: true
    peer_closed_channel:   true
    blocks_since_close:    42184
    capacity:              200000
    channel_id:            642993x877x1
    channel_open:          53f3218c2eea32f40801d55ddc245106a3417e32ed0980280487ab94e8ffe24e:1
    channel_close:         5aaeba79258f87b56319750d7ba533e0a2fa1d29bb03840417adfe1f163f9aa0
    channel_resolutions: 
      - 
        value:          186340
        transaction_id: 7d1c2b2d54ab6a8ae547e814166c464e25cea0fa77843ae4ebe6777112030e93
        type:           p2wpkh
  - 
    peer_public_key:       02360f8e4f4181892136af6fe302c4017fb1e57672505e8584ae2acac87792deb6
    peer_alias:            fishfeed
    is_remote_force_close: true
    peer_closed_channel:   true
    blocks_since_close:    42180
    capacity:              200000
    channel_id:            642983x2096x0
    channel_open:          e262af2d7cd9db49b76989d4ad12c2ad9eef7d4bb74d0735a73e1bb08776fa38:0
    channel_close:         457982ffb88e5b4cd2d1f0f32164153bb9e561550cbabbfe8fcee8905a15382c
    channel_resolutions: 
      - 
        value:          186388
        transaction_id: 14217657b6b8b46b7af7a954be8764255d32cbb30aa541303173e7b9ae52a480
        type:           csv_delayed
  - 
    peer_public_key:      03fab7f8655169ea77d9691d4bd359e97782cb6177a6f76383994ed9c262af97a5
    peer_alias:           🐸webworker.sh🐸
    is_cooperative_close: true
    blocks_since_close:   30984
    capacity:             93371
    channel_id:           665646x1539x1
    channel_open:         87d20c0b37177d878d37f77e23b8df355e9989e03f0189ee622a978207cd0f41:1
    channel_close:        9068cd09545baa4626a23c8aa8a987b47e7edd8494fa5db550c73fee18a1c588
    closing_fee_paid:     23304
  - 
    peer_public_key:      020ec1b1be32061d3c5264be7e6d63a59a28b89e06eb7adc619590adabb080469d
    is_cooperative_close: true
    blocks_since_close:   30675
    capacity:             30000
    channel_id:           666378x157x0
    channel_open:         f89018a6a0faebec4b6a9dfbe7ed95e7bacdf8e4156231d9711196b2c972c72f:0
    channel_close:        75f1893661a086be40497e1df526b666e77b35f27fb86e11e2717f708fda505b
    closing_fee_paid:     14999
  - 
    peer_public_key:      03f57da433d006dd13446b760721c7e9d13bdc8e4f9c1bc402f30084833f21c4dd
    peer_alias:           03f57da433d006dd1344
    is_cooperative_close: true
    peer_closed_channel:  true
    blocks_since_close:   19163
    capacity:             30000
    channel_id:           666799x1477x1
    channel_open:         b60f8acc614d3284fa6a633e57057f7312c1f476b9c55f1eba3fb3c59d755147:1
    channel_close:        4f63d61667b1fe6ded2dd5501bbfa134c3003cbadb61176b727b17df9eee6295
    closing_fee_paid:     6623
  - 
    peer_public_key:      02feb7c345ee51c6f1c4eca60296ead1e347db4a2fd8b981dde55f39c0409a291f
    peer_alias:           02feb7c345ee51c6f1c4
    is_cooperative_close: true
    blocks_since_close:   4860
    capacity:             300000
    channel_id:           671177x1847x1
    channel_open:         1b460135c4acc5e0d9daecbc4856bc8747090f80d883b14815b35dfb2f883e93:1
    channel_close:        35578e645d0c9416d2f01f255e35a55c37e6f6b45c1f9b4e7ca9afb017d4f2be
    closing_fee_paid:     366
  - 
    peer_public_key:      02e26f050f82c63e8895938ce782bb123de9580a3e1100531f6ec1cdfc7c25fe2f
    peer_alias:           mynodebtc.com [KiloBravo1]
    is_cooperative_close: true
    blocks_since_close:   4746
    capacity:             30000
    channel_id:           666377x638x0
    channel_open:         ae069a3023b6ea6fe7877e311316c646086d56edefea94516d426d93e6ab8db1:0
    channel_close:        23880ae06a311dd33a7dedcbd25abed80e9f4def6723e30134da29461eb0639a
    closing_fee_paid:     348
  - 
    peer_public_key:      02c87ddac0fb9ddbcc88557275419dd9a307275eabe9d2f74e338d90b0eda81be7
    peer_alias:           IrreduxMX
    is_cooperative_close: true
    blocks_since_close:   4464
    capacity:             25000
    channel_id:           666362x2765x0
    channel_open:         3880984f66e6a876fd85ffa6b41b760ebfc19b3ec5ba15491f00e126e52de918:0
    channel_close:        65dc11148cbb8fb1cb912be6ad58ed0558a52f45d27374951c9919b6fd5bd20d
    closing_fee_paid:     260
`

clockshell, 2021-09-04:
> I presume 372168 sats should have hit your wallet a while ago https://blockstream.info/address/bc1q9j9zpxtzywv8ly98s7s5s853nqtuxm4ym40pex

I just re-checked again. I cannot find a transaction with that amount.

rootzoll, 2021-09-09:
From the RaspiBlitz perspective there seems no error (closing issue). If this an issue with LND (not displaying/catching some transactions) .. please pitch your question over to their community support channels: https://dev.lightning.community/

closed 2021-09-09 by rootzoll