raspiblitz/raspiblitz #2453: Copy blockchain over LAN failing. Possible due to lack of permissions

tifDev, 2021-07-31:
I want to copy the blockchain from one working node to a new one. Both are under the same network.
I'm getting this error:

```

# Starting copy over LAN (around 4-6 hours) ...
sending incremental file list
copy_begin.time
rsync: mkstemp "/mnt/hdd/.bitcoin.rLpXFy" failed: Permission denied (13)

sent 126 bytes  received 112 bytes  158.67 bytes/sec
total size is 11  speedup is 0.05
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
rm: cannot remove './transferred.rsync': Permission denied
sending incremental file list
rsync: mkdir "/mnt/hdd/bitcoin" failed: Permission denied (13)
rsync error: error in file IO (code 11) at main.c(682) [Receiver=3.1.3]

```
Seems that this is being caused by permission issues. Can you help fixing this?

**Note:** The working node (source node) has an HDD 1tb and the new node (target node) has an SSD 1tb, not sure if this can also cause conflicts.

Thank you

nyxnor, 2021-08-02:
```
ls -l /mnt/hdd/bitcoin
ls -la /mnt/hdd/.bitcoin
```

xanoni, 2021-08-04:
The problem is most likely that the remote folder `/mnt/hdd/bitcoin` does not exist yet and the `bitcoin` user doesn't have permission to create new folders in `/mnt/hdd`. This is likely fixed by `sudo mkdir /mnt/hdd/bitcoin && sudo chown bitcoin:bitcoin /mnt/hdd/bitcoin` on the remote machine.

Alternative:

https://github.com/rootzoll/raspiblitz/blob/fc1d44104bc22d85ebebd7b0d6ee2240722a9edf/home.admin/98repairMenu.sh#L114

Use the `root` user to log in instead, or use `admin` with parameter `--rsync-path 'sudo rsync'`.

tifDev, 2021-08-09:
I wiped out my sd card and SSD and now I'm not able to reproduce it anymore. I will reopen this issue if it happens again, closing for now...

Thank you all.

closed 2021-08-09 by tifDev