raspiblitz/raspiblitz #2454: BTRFS build fails with RaspiKey

STEMr42, 2021-07-31:
rPi 4 with 32GB USB and RaspiKey eMMC module for OS fails the status check for BTRFS.

https://www.uugear.com/product/raspikey-plug-and-play-emmc-module-for-raspberry-pi/

lsblk lists 2 additional boot devices causing raidCandidates=3

`mmcblk0boot0 179:32   0     4M  1 disk`
`mmcblk0boot1 179:64   0     4M  1 disk `

These pass the criteria set for raid candidates in blitz.datadrive.sh line 397 as they are listed as "disk" and do not contain "/" (couldn't this just search for sd* that is not main hdd?).

Failing section:

`    for disk in $(lsblk -o NAME,TYPE | grep "disk" | awk '$1=$1' | cut -d " " -f 1)`
`    do`
`      devMounted=$(lsblk -o MOUNTPOINT,NAME | grep "$disk" | grep -c "^/")`
`      # is raid candidate when not mounted and not the data drive cadidate (hdd/ssd)`
`      if [ ${devMounted} -eq 0 ] && [ "${disk}" != "${hdd}" ] && [ "${hdd}" != "" ]; then`

An additional check is needed to ignore them:

`    for disk in $(lsblk -o NAME,TYPE | grep "disk" | awk '$1=$1' | cut -d " " -f 1)`
`    do`
`      devMounted=$(lsblk -o MOUNTPOINT,NAME | grep "$disk" | grep -c "^/")`
`      devBoot=$(lsblk -o NAME | grep "$disk" | grep -c boot)`
`      # is raid candidate when not mounted and not the data drive cadidate (hdd/ssd)`
`      if [ ${devMounted} -eq 0 ] && [ "${disk}" != "${hdd}" ] && [ "${hdd}" != "" ] && [ ${devBoot} -eq 0 ]; then`

xanoni, 2021-08-04:
Just out of curiosity, why did you choose BTFRS?

For the other curious folks, this summarizes BTFRS: https://wiki.archlinux.org/title/Btrfs

STEMr42, 2021-08-22:
> Just out of curiosity, why did you choose BTRFS?
> 
> For the other curious folks, this summarizes BTRFS: https://wiki.archlinux.org/title/Btrfs

It was already in the build, just didn't work with the emmc modules.

rootzoll, 2021-08-23:
Just for quick background why BTRFS and not ZFS (for now): If you search the issues (also closed) for ZFS you should find some discussions on that already. The one that at let me keep my hands off ZFS for RaspiBlitz was the following:

Yes ZFS on RaspberryPi even with low memory configs are possible - you can find tutorials that show ZFS on RaspberryPi. But ZFS for ARM is not common - thats why for a TruNAS you normally need a x86. If we would switch to ZFS with RaspiBlitz we would most prob become one of the biggest projects on ZFS build for ARM processors. And thats a very shaky ground to stand on - if we run into a bug down that lane there is a big change to get stuck.

The chapter on ZFS is not closed ... if there is another big RaspberryPi project running ZFS with no problems for a good long time we might reconsider.

rootzoll, 2024-11-14:
Also BTRFS might get removed from RaspiBlitz now - https://github.com/raspiblitz/raspiblitz/issues/2045#issuecomment-2471845122

The idea is when you want to use BTFS or ZFS you should consider to run RaspiBlitz in a VM like with Proxmox.

closed 2024-11-14 by rootzoll