raspiblitz/raspiblitz #2455: [lncli] rpc error: code = unknown desc = no policy for outgoing channel 762664146008014849

ronledfordjr, 2021-08-01:
[image]

RPi 4 8GB
ver 1.7.0
LND 0.12.1-beta

I cloned my 16GB Mciro SD to a 32GB. RPi 4 started up fine and seemed to me working as expected.

I opened a Triangle Swap on Lightning+

I have copied in the debug report because the shortcut cutoff a substantial part of report.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.0
chainnetwork: bitcoin / main
 02:27:24 up 33 min,  2 users,  load average: 3.76, 3.66, 3.18

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-08-01 01:29:16 BST; 58min ago
  Process: 1597 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1601 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1635 (bitcoind)
    Tasks: 18 (limit: 4915)
   CGroup: /system.slice/bitcoind.service
           └─1635 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/…

Aug 01 01:29:15 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 01 01:29:16 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2021-08-01 01:28:17 BST, end at Sun 2021-08-01 02:27:24 BST. --
Aug 01 01:29:15 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 01 01:29:16 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-08-01T01:20:51Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable
2021-08-01T01:20:56Z Socks5() connect to wlhdbyk2fm5ugp4z5be6zxyvkgp6pza5w6wkzoa4gj3d2ds2swhz3zyd.onion:8333 failed: host unreachable
2021-08-01T01:21:57Z New outbound peer connected: version: 70016, blocks=693639, peer=39 (full-relay)
2021-08-01T01:22:05Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2021-08-01T01:22:17Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2021-08-01T01:22:29Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2021-08-01T01:22:40Z Socks5() connect to ldvhlpsrvspquqnl3gutz7grfu5lb3m2dgnezpl3tlkxgpoiw2g5mzid.onion:8333 failed: host unreachable
2021-08-01T01:22:47Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable
2021-08-01T01:22:52Z UpdateTip: new best=000000000000000000066902699d281a756ed716a31d85a4b6d7a2f80c9d72ec height=693640 version=0x37ffe004 log2_work=93.000372 tx=659445058 date='2021-08-01T01:22:32Z' progress=1.000000 cache=4.5MiB(33125txo)
2021-08-01T01:22:52Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2021-08-01T01:24:11Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2021-08-01T01:24:24Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2021-08-01T01:24:35Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2021-08-01T01:24:52Z Socks5() connect to ldvhlpsrvspquqnl3gutz7grfu5lb3m2dgnezpl3tlkxgpoiw2g5mzid.onion:8333 failed: host unreachable
2021-08-01T01:24:56Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable
2021-08-01T01:26:26Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2021-08-01T01:26:41Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2021-08-01T01:26:51Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2021-08-01T01:27:02Z Socks5() connect to ldvhlpsrvspquqnl3gutz7grfu5lb3m2dgnezpl3tlkxgpoiw2g5mzid.onion:8333 failed: host unreachable
2021-08-01T01:27:07Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-08-01 01:29:16 BST; 58min ago
  Process: 1637 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 1644 (lnd)
    Tasks: 12 (limit: 4915)
   CGroup: /system.slice/lnd.service
[... 275 lines cut ...]
*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.198'
tempCelsius='58.4'
uptime=2016
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=693640
initialSync=0
syncProgress=100.00 
startcountLightning=1
lndActive=1
walletLocked=0
lndPeers=4
syncedToChain=1
scanTimestamp=1627780952
scanDate='Sun  1 Aug 02:22:32 BST 2021'
scanProgress=99.99
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=4

nyxnor, 2021-08-02:
Could you open this issue on LND github? https://github.com/lightningnetwork/lnd/issues/2575

xanoni, 2021-08-03:
A bit random, but I noticed that `lnbits` is listening on `http://0.0.0.0:5000` ... should probably be using SSL if it is listening on all interfaces?

xanoni, 2021-08-03:
By the way, I looked up the channel on 1ML and it was closed yesterday (2021-08-01 15:22:52 GMT -4). I assume this ticket is no longer needed then.

https://blockstream.info/tx/c36bbae85156eb8ea85514c26707b70aace41e09568629624b5286e862ceb3ea

https://1ml.com/channel/762664146008014849

xanoni, 2021-08-04:
By the way, for future upgrades, this document explains how to properly migrate your `lnd` to new hardware:

https://docs.lightning.engineering/lightning-network-tools/lnd/safety#migrating-a-node-to-a-new-device

openoms, 2021-08-10:
Do not clone the SDcard. Flash a clean image and let the recovery process take it's course.

ronledfordjr, 2021-11-02:
It is running now.

closed 2021-11-02 by ronledfordjr