raspiblitz/raspiblitz #2457: Tweaking `nice`ness in systemd scripts

xanoni, 2021-08-02:
I noticed that a ton of stuff is competing for CPU on my Pis, my average load is typically 5-10 [1]. Thus, I have added `Nice=X` lines to my `systemd` config scripts in an attempt to make things feel more snappy.

I'm still testing, but it seems that giving lower priority (=positive nice values) to the always on background stuff (e.g., `lnd` and `bitcoind`) and higher priority (=negative nice values) to the frontend-serving processes (e.g., `btcrpcexplorer`, `LndHub`, `sshd`, ...) could help.

[1] This might be higher than a Raspiblitz, given I also run `monerod`, have an always-on rebalancing & fee update script, as well as LUKS full-disk encryption.

vv01f, 2021-08-05:
I tried `renice -19` on my syncing bitcoind recently, didnt seem to make much difference; but can be misleading as I was comparing different periods of the blockchain. This may be worth a test with two separate sets and comparing when on LAN with a dedicated source node. its a bit effort on top to keep situations comparable.

xanoni, 2021-08-05:
I agree that it's difficult to benchmark this, as it will be difficult to consistently retest a realistic scenario. Also, some of what we're trying to achieve (responsiveness) may be difficult to benchmark in the first place.

I'll keep my settings, even though I can't claim that there's a _measurable_ improvement.

xanoni, 2021-08-12:
By the way, playing with the `ionice` values probably does even more ... default is this:

```
IOSchedulingClass=best-effort
IOSchedulingPriority=0
```

See `man ionice`.

xanoni, 2021-08-18:
Another nice resource management tool in our toolkit is `taskset` ... e.g., the below command would ensure that the launched command would only use the third and fourth CPUs, but leave CPUs 1 and 2 alone:

```bash
# can be used in systemd scripts
tasksel --cpu-list 3,4 ./some_daemon_to_be_launched
```

xanoni, 2021-08-23:
For completeness, I also wanted to mention all the `LimitX=` items that we can tweak.

See: https://www.freedesktop.org/software/systemd/man/systemd.exec.html#Process%20Properties

E.g., I've sometimes seen `LimitNOFILE=$MAX_FILE_DESCRIPTORS` (~`ulimit -n`) used in the Bitcoin context.

xanoni, 2021-08-24:
Found more interesting stuff ... treating this as my notepad ...

https://www.freedesktop.org/software/systemd/man/systemd.resource-control.html

**_Mostly performance:_**
- `AllowedCPUs=` (looks like the proper way to achieve what `taskset` is doing, see my previous comment)
- `MemoryMax=`
- ` CPUQuota=`
- `MemoryHigh=bytes`
- `IOWeight=weight, StartupIOWeight=weight`
- `IODeviceWeight=device weight`
- `IOReadBandwidthMax=device bytes, IOWriteBandwidthMax=device bytes`
- `IOReadIOPSMax=device IOPS, IOWriteIOPSMax=device IOPS`

**_Mostly security:_**
- `IPAddressAllow=ADDRESS[/PREFIXLENGTH]…, IPAddressDeny=ADDRESS[/PREFIXLENGTH]…`
- `SocketBindAllow=bind-rule, SocketBindDeny=bind-rule` (e.g., `SocketBindAllow=ipv4:udp:10000-65535`, `SocketBindDeny=any`)
- `DeviceAllow=` (protect HW wallets?)