raspiblitz/raspiblitz #2458: Cannot unlock LND wallet. Repeatedly being asked for password C

zorke1, 2021-08-02:
Hi,

Please help me with explanation for dummies. It keeps asking for password C. I have flashed the SD card but no sucess. I do not know what to do. Thanks

/home/admin/XXdebugLogs.sh | nc termbin.com 9999

zorke1, 2021-08-02:
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.7.0
chainnetwork: bitcoin / main

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.7.0
chainnetwork: bitcoin / main
 20:02:36 up 38 min,  2 users,  load average: 1.55, 1.44, 1.28

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-08-02 18:36:22 BST; 1h 26min ago
  Process: 1541 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1547 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1568 (bitcoind)
    Tasks: 12 (limit: 4533)
   CGroup: /system.slice/bitcoind.service
           └─1568 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitco…

Aug 02 18:36:22 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 02 18:36:22 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2021-08-02 18:35:27 BST, end at Mon 2021-08-02 20:02:36 BST. --
Aug 02 18:36:22 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 02 18:36:22 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-08-02T19:02:12Z Rolling forward 0000000000000000000eaebdb752c80e7acf6372f8f28a776448f42c2f7c4b95 (614647)
2021-08-02T19:02:13Z Rolling forward 00000000000000000001a5f99008f02705fa5ff9164cab2fd541e9901fbf792c (614648)
2021-08-02T19:02:13Z Rolling forward 0000000000000000000e73baafe09e55e1c0fb38ae0de4940bcf6cdb29beb99e (614649)
2021-08-02T19:02:15Z Rolling forward 0000000000000000000f0aaf808888aac89b7101bcea45e023124027f027b4da (614650)
2021-08-02T19:02:16Z Rolling forward 000000000000000000094009939a1197b93ffe173609081c3e367da250c90a62 (614651)
2021-08-02T19:02:17Z Rolling forward 00000000000000000011871bc6fcc926a54a73b829d6c52dde23a7197c0dd4b9 (614652)
2021-08-02T19:02:19Z Rolling forward 000000000000000000072f30253c0ed0b7012748dbdb2a56ef3e894c9a05e619 (614653)
2021-08-02T19:02:20Z Rolling forward 0000000000000000000b5bd42a29f322d15fa1d5c78fe726179be76bb1c63c08 (614654)
2021-08-02T19:02:22Z Rolling forward 0000000000000000000f86d29fc3bf67c9d15ba5afcb15a7f562c7684faa4959 (614655)
2021-08-02T19:02:23Z Rolling forward 0000000000000000000fbb7b09ea3c7c26bf6ed3146b0e26c183b86705c58c45 (614656)
2021-08-02T19:02:24Z Rolling forward 00000000000000000007e1f2651414a76d10e4a0a1ca96ae3f98f54fa0969bf2 (614657)
2021-08-02T19:02:24Z Rolling forward 00000000000000000012855674a11155c23debb3f10d9c2d39e6228f12f3eac3 (614658)
2021-08-02T19:02:25Z Rolling forward 0000000000000000000ca92cc9973be6025f2533e0818e014a457fb6aba5753e (614659)
2021-08-02T19:02:26Z Rolling forward 0000000000000000000b297e100831e2ea8b092d8b1cc121b4e7e2b68ce83b88 (614660)
2021-08-02T19:02:27Z Rolling forward 0000000000000000000092c11e184ab285d46e61fb406807925685e22ee9827e (614661)
2021-08-02T19:02:28Z Rolling forward 0000000000000000000dbdd23f8d958899455f1809d041bdd0670272e568dad1 (614662)
2021-08-02T19:02:29Z Rolling forward 00000000000000000007e8cd624ac1c77d26af008f5b2ac2cd2b49d50cf3a83b (614663)
2021-08-02T19:02:32Z Rolling forward 00000000000000000008d0fa5cf73053f8e1d3da25a6a61330319250fc203969 (614664)
2021-08-02T19:02:34Z Rolling forward 00000000000000000001f58abe02c8a8f7b75cf3bf8f2228198932dbc05584d4 (614665)
2021-08-02T19:02:36Z Rolling forward 0000000000000000001019f7f1f8ce91dd97209dac81589e864e7b3ccb4a4558 (614666)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2021-08-02 20:02:33 BST; 3s ago
  Process: 1465 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 1467 ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --tor.socks=9070 --tor.control=9071 --listen=127.0.0.1:9735 ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 1467 (code=exited, status=1/FAILURE)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Mon 2021-08-02 18:35:27 BST, end at Mon 2021-08-02 20:02:36 BST. --
Aug 02 19:29:12 raspberrypi systemd[1]: Started LND Lightning Daemon.
Aug 02 19:58:52 raspberrypi lnd[7136]: unable to create chain control: unable to connect to bitcoind: -28: Loading block index...
Aug 02 19:58:52 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
[... 189 lines cut ...]
dhcp=1
network_device=eth0

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.20'
tempCelsius='54.5'
uptime=2319
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
bitcoinErrorShort='Loading block index'
bitcoinErrorFull='error code: -28 error message: Loading block index...'
startcountLightning=3
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

stevenhjxu, 2021-08-03:
I am in the same situation. Previously it was showing the download % on the screen, but all of sudden it was giving the message of "bitcoind is not running". With the research I've done, and judging from your log, it looks like you are loading block index, and the only solution is to wait. It looks like though you are loading at 1 per second, which is pretty good. Mine is loading at 1 per 6 seconds, and I have a long way to go. I've been downloading the blockchain for 9 days now, and quite frustrated with how slow it is...

zorke1, 2021-08-03:
Hi,
Thanks for leaving your comment. Thing is I am not sure waiting will do anything good as I am being asked for password C repeatedly. If I do not input it, nothing happens. It is just a loop as I can tell. Asks for pw C I put it in, LND gets unlocked and after verifying it asks for pw C again and it keeps looping like this.

I don't know, I hope someone that understands the issue will respond and she'd some light on a solution as I have hit the brick wall.

stevenhjxu, 2021-08-03:
What I have been doing is to cancel when it's asking for password, then before it asking again, ctrl c out of it, then type menu, and you can go into System Options -> bitcoinLOG to monitor the block indexing. You could give it a try if you want, your debug log is very similar to mine by the way.

zorke1, 2021-08-03:
I have done it. SO are you trying to say that BTC blockchain is actually being downloaded and node is updating? What about LND node, is it updating? If everything is being updated, how will we access our LND wallet when pw C is not recognized? And I am also wondering what would happen if we would go and do RESET LND option in REPAIR? I do not mind seting up new LND wallet and passwords if this would help fix my issue. Too many questions my friend :)

stevenhjxu, 2021-08-03:
I actually just started dabbling with it, so I don't know enough to answer your questions. From all the googling I've done, it is the conclusion that it's trying to download the Bitcoin blockchain. I think the total block number is 681K, so you are about 90% there. Regarding the lightning network, from the log, it seems the password is recognized, but since the Bitcoin network is not running, it just forces quit the wallet. But that's my guess though.

zorke1, 2021-08-03:
Yes it seems we need to wait for node to finish updating:

https://github.com/rootzoll/raspiblitz/issues/2398#issuecomment-873890920

Let's see. All the best!

rootzoll, 2021-08-17:
FYI with v1.8 we will change the wallet locked detection to the new LND RPC status - see #2470

rootzoll, 2021-09-09:
v1.7.1 is comming up soon .. that already has some improvements in the wallet unlock. With v1.8 we will then use the advanded LND options on wallet unlock detection.

openoms, 2021-10-06:
@felivix 

For now if you use autounlock disable it.
you can enter the menu by typing:
`menu`

Then in the command line again you can unlock your wallet with:
`lncli unlock`
This has to be done after every restart until the v1.7.1 update.

felivix, 2021-10-06:
@openoms

lncli unlock does its job, until the wallet auto-locks again after a few minutes.

Auto-locking of the LND Wallet happens consistently all the time (a few minutes after unlocking it).

openoms, 2021-10-06:
In that case you must have some error unrelated to the unlock change.

Use the `menu` -  SYSTEM` - `LNDLOGS` to tail the logs or look out for errors in:

`sudo journalctl -fu lnd`