raspiblitz/raspiblitz #2460: addming xmpp/jabber for notifications

vv01f, 2021-08-03:
concerns #2231

I added XMPP with [sendxmpp](https://www.systutorials.com/docs/linux/man/1-sendxmpp/) for my own needs locally:

1. `sudo apt install sendxmpp -y`

2. wrote `~/sendxmpprc`:

```
# echo "Message" | sendxmpp -u user -p pass -j domain.tld -n receipient@domain.tld
username: user
jserver: domain.tld
password: pass
# port: 5222 # default is 5222
tls: 1 # modern servers will not accept anything without TLS
# verify: 1 # unknown value for off, use cli paramater -n if needed
```
3. added command

```
# command: xmpp
# send message via xmpp to standard receipent with user from settings
xmpp ()
{
  grep "notify=on" /mnt/hdd/raspiblitz.conf || { echo "notify is inactive"; return; } # assert when disabled
  # grep -i "notifyMethod=xmpp" /mnt/hdd/raspiblitz.conf || { echo "Method is not XMPP"; return; } # assert when not xmpp
  r=$(cat /mnt/hdd/raspiblitz.conf|grep -i notifyXMPPTo|cut -d"=" -f2) # fetch receipient
  echo -e $(date +'%Y-%m-%d %H:%M %z')"\n"$(whoami)"@"$(hostname)":\n\n$@" | sendxmpp -n "${r}" # send message
  test $? -eq 0 || { echo "error on sending XMPP message."; return 1; }
}
```
…

nyxnor, 2021-08-07:
Dont you need to host a xmpp server on blitz to send the notification?
I have a script that set up prosody with onion module and otr.

raucao, 2021-08-09:
@nyxnor Why would it have to be a local server? You will need an account for the XMPP client/script to send notifications from either way, but it shouldn't matter where the XMPP server is running.

vv01f, 2021-08-09:
@nyxnor what was your alternatives supporting OTR/OMEMO again? That encryption would be neat as the current tool would be cleartext only with the option to send inline PGP where I doubt receiving clients will be able to handle it.

nyxnor, 2021-08-09:
OMEMO - https://github.com/gkdr/lurch
OTR - https://github.com/dgoulet/prosody-otr/master/mod_otr.lua
ONIONS - https://hg.prosody.im/prosody-modules -> mod_onions/mod_onions.lua

nyxnor, 2022-01-17:
sendxmpp does not support OTR.
https://github.com/lhost/sendxmpp/issues/8

but the issue last comment is from 2014 and maybe has been updated so far.

so the receiving server will be able to read the messages sent... I will see if it is possible to work around this issue.

nyxnor, 2022-01-17:
or if someone knows per to implement it https://github.com/ajgb/protocol-otr/blob/master/examples/xmpp-otr.pl

raucao, 2022-01-17:
I think the more appropriate method of encryption for this kind of use case would be PGP. See e.g. https://xmpp.org/extensions/xep-0027.html or https://xmpp.org/extensions/xep-0373.html (the former being more widely implemented in current clients than the much more recent new spec).