raspiblitz/raspiblitz #2465: supporting BIP 157

vv01f, 2021-08-05:
for enabling connection from e.g. Breez, see https://github.com/breez/breezmobile/wiki/Connecting-a-Bitcoin-node-supporting-BIP-157

* [x] bitcoind to 0.21 (in raspiBlitzVersion 1.7.0)
* [ ] add to conf `blockfilterindex=1` and `peerblockfilters=1`

is there any conflict with current configuration details like `peerbloomfilters`?

nyxnor, 2021-08-07:
I think Bisq requires peerbloomfilters=1

openoms, 2021-08-10:
yes, Bisq needs peerbloomfilters, but there is no conflict with peerblockfilters.
you can edit your bitcoin.conf from the SYSTEM menu.
The connection will be over LAN only as Breez does not have Tor support bulit-in. You can experiment with Orbot on Android, but it is unlikely to be reliable.

vv01f, 2021-08-10:
> yes, Bisq needs peerbloomfilters, but there is no conflict with peerblockfilters.

thx :) 

> The connection will be over LAN only as Breez does not have Tor support bulit-in.

yep, but it's on roys roadmap, so we might add it to the connect wallets after that is working. :)

rookey, 2024-12-12:
Is this supposed to work already? I applied the changes to bitcoind config, but when trying to connect to my Raspi's IP, Breez still responds "Breez is unable to connect to the specified node. Please make sure this node supports BIP 157."