raspiblitz/raspiblitz #2466: Raspiblitz keeps launching the menu "Copy blockchain from another latptop/node over lan"

tifDev, 2021-08-08:
So I started the copy of the blockchain from my source node to the target node using my laptop to connect to them and launch the commands.

After some time my laptop lost the ssh connection to the RPI's due to idle time.

I connected to the target RPI to check if the copy went well and it launched this menu:

[image]

Then I select BLITZ > Keep the blockchain data > then launches this screen:

[image]

I pressed enter the says that the data is good and restarts.

When I connect again it repeats the same process every time, I'm stuck.

**How can I solve this?**

**NOTE**: The first time I connected to the target RPI after the copy of the blockchain finished it asked also to set Password C to lightning, then created a new Lightning wallet and restarted automatically.

openoms, 2021-08-10:
I would suggest to restart the process by flashing a new SDcard.

Make sure the bitcoin node is stopped on the source node.

tifDev, 2021-08-11:
I wanted to avoid doing the whole process again.

rootzoll, 2021-08-16:
The v1.7.1 will come with a refactored setup process - I hope this will fix such situations. For now I dont think there is a quick fix other then try a fresh sd card and if still stuck format SSD fresh and start sync from the beginning.

tifDev, 2021-08-18:
In other setup it also gave errors but in the other day it worked and now it's fine. The blockchain copy over Lan it's a bit unstable but yes, let's hope that it will get fixed with next release.

Taking the moment to ask you if I can copy from a PC by connecting both disks in the same computer? https://github.com/rootzoll/raspiblitz/issues/2484

Thank you.

closed 2021-08-18 by tifDev