raspiblitz/raspiblitz #2467: Tor: "Failed to start Anonymizing overlay network for TCP"

raucao, 2021-08-08:
My node's default tor instance is failing to (re)start after reboots with the following error in the service logs:

```plain
-- The unit tor@default.service has entered the 'failed' state with result 'exit-code'.                                                                                     
Aug 08 13:05:34 raspberrypi systemd[1]: Failed to start Anonymizing overlay network for TCP. 
```

The lnd tor instance works.

When trying to start tor manually via the exec start command, there's a warning for the web80 hidden service directory's permissions being too permissive, as well as this error message:

> Failed to parse/validate config: Failed to configure rendezvous options.

Unfortunately, I couldn't find useful information about that error yet. However, most discussions I found on the Web were about issues with running tor as a certain user. So perhaps it has something to do with trying to start a hidden service on port 80 as "debian-tor"?

raucao, 2021-08-08:
As a pointer to what the problem might be: I have manually disabled the hidden services in `/etc/tor/torrc` for now, which does fix the Tor startup issue so that the service can run again.

nyxnor, 2021-08-09:
```
sudo chown -R debian-tor:debian-tor /mnt/hdd/tor
```

raucao, 2021-08-09:
Thanks! I should have added to the original post that I already verified the ownership of the files in that directory.

openoms, 2021-08-10:
The fix in case of manual changes is to reflash the SDcard to get back to the tested state.

raucao, 2021-08-10:
I have not done any manual changes to the Tor config or anything related before it failed. If you read my issue description carefully, you can see that I only did a manual config change to hotfix the problem when it started failing after reboots.

I understand that you don't want to help debug problems when people mess around with their devices. But if you don't discern between broken by user and broken by itself, and never check to see what's what, then that's probably one of the reasons why everything is so fragile. Re-flashing the whole thing should be a last-resort solution in case of dramatic issues in my opinion, not something you casually do whenever there's an issue on a running system.

nyxnor, 2021-08-10:
I have an open PR for this. I had the same problem and do not know why that happened.
https://github.com/rootzoll/raspiblitz/pull/2322

raucao, 2021-08-10:
The result is indeed the same (including the main GUI script hanging at 99% sync when starting after the reboot). However, for me it wasn't the ownership of the directory, which is correctly set to `debian-tor`, but instead a warning about the directory mode being too permissive. The hidden service dirs on my device are set to 720 atm:

```
-rw-rw---- 1 debian-tor debian-tor 937234496 Aug 10 13:00 info.log
...
drwx-w---- 1 debian-tor debian-tor       136 Jul  3 19:24 lndrest8080
drwx-w---- 1 debian-tor debian-tor       136 Jul  3 19:24 lndrpc10009
-rw-rw---- 1 debian-tor debian-tor   1081348 Aug 10 12:59 notice.log
...
drwx------ 1 debian-tor debian-tor       192 Aug 10 13:00 sys
drwx-w---- 1 debian-tor debian-tor       136 Aug  8 13:45 web80
```

openoms, 2021-08-10:
The specific users, directories and permissions are set up by Tor itself, we are mainly working around it trying to reduce the problems as much as possible.
Starting tor manually with the relevant user and looking at the output is the way to go.
Most of the time needs a deep dive to the systemd service files (set up by Tor).

openoms, 2021-08-10:
Sorry for the premature close. Feel free to close when resolved or if there is nothing actionable in this repo.

nyxnor, 2021-08-13:
Before restarting tor, I am doing this https://github.com/nyxnor/raspiblitz/blob/becc06be71440eb9360d47a602bf1868242e821e/home.admin/config.scripts/tor.functions.lib#L99
on pr #2419 
https://github.com/nyxnor/raspiblitz/blob/becc06be71440eb9360d47a602bf1868242e821e/home.admin/config.scripts/tor.functions.lib#L176

```bash
set_owner_permission(){
  sudo chown -R ${OWNER_DATA_DIR}:${OWNER_DATA_DIR} ${DATA_DIR}
  sudo chown ${OWNER_CONF_DIR}:${OWNER_CONF_DIR} ${TORRC}
  sudo chmod 700 ${DATA_DIR}
  sudo chmod 644 ${TORRC}
}
```

raucao, 2021-08-29:
I forgot to report back here. The problem had indeed been caused by wrong permissions. The default permissions, and the ones I have set the directories (back?) to, are actually `0700` for the hidden service folders and `0600` for the files in them (not `0644`, as used by @nyxnor). This makes sense to me, as only the Tor user is able to read the private keys then.

Closing the issue, because it's unlikely I can find out what changed the permissions at this point. But I hope it's useful for someone else potentially running into it, or for a future fix.

closed 2021-08-29 by raucao