raspiblitz/raspiblitz #2472: FAIL on RTL install

kayloehmann, 2021-08-11:
Hi everyone,

i just downloaded and setup a raspiblitz today.
When trying to enable RTL service i get this error.

[image]

Following the proposed solution does not have any effect.

What can i do?

Thanks!

kayloehmann, 2021-08-11:
I fixed it by installing a new image!

closed 2021-08-11 by kayloehmann

ugligatschu, 2021-09-28:
Hi, i have the same problem, what was your solution? Greets ugligatschu

rootzoll, 2021-09-28:
Have you run the command (as shown in the message) on the terminal manually to see the error during install? 

BTW new RaspiBlitz v1.7.1 will be released beginning of October. RTL install will be updated and tested in that version ... so a simple solution would be to wait a bit and then update.

ugligatschu, 2021-09-28:
Yes i tried it like this:
```
admin@edisraspiblitz:~ $ /home/admin/config.scripts/bonus.rtl.sh on
# making sure services are not running
# INSTALL RTL
nodeJS is already installed
Installed nodeJS v10.24.0
adduser: Der Benutzer »rtl« existiert bereits.
```

ugligatschu, 2021-09-28:
and btw, newer go asked to set "password D" during installation?

rootzoll, 2021-09-28:
Hmmm .. something get messed up on your old RaspiBlitz. Best would be to insert a fresh sd card and make a fresh recover. 

About "Password D" ... password D is deprecated - its no longer needed. RaspiBlitz works now just with Password A,B & C.

ugligatschu, 2021-09-29:
Ok thanks i will do this. 
where do i find how to prepare a „fresh recovery“. Does the thos impact open channels?

ugligatschu, 2021-09-29:
I got some help and it seems to work again.
`sudo userdel rtl `
And then 
`/home/admin/config.scripts/bonus.rtl.sh on`
worked for me