raspiblitz/raspiblitz #2477: unexpected config eof on power loss

nicolasdiehl, 2021-08-13:
Recently I wanted to connect via ssh but couldn't, having no display and no funds on it yet I tried restarting with a power reset to see what happened, this happened:

Starting the main menu ...
/dev/fd/63: line 26: unexpected EOF while looking for matching `''
/dev/fd/63: line 28: syntax error: unexpected end of file
loading config data
checking bitcoind - please wait ..
can take longer if device was off or first time

these errors appear every time the status screen updates. I'll do some debugging later but in the meantime you could give me some advice. Thanks.

rootzoll, 2021-08-16:
The power reset seems to have corrupted some files .. best procedure here to start with a fresh sd card and let the system recover.

nicolasdiehl, 2021-08-17:
I connect with ssh. I found out that the errors only appear when the function is called within another py file. if i call it directly then there are no errors. could it be that there are some minor piping issues going on with displaying stuff on ssh? will update later with exact location and outputs

rootzoll, 2021-08-18:
Can you be more detailed what py-scripts you are calling?

nicolasdiehl, 2021-08-22:
The chain goes like this:
https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/00raspiblitz.sh#L13 calls
`source <(sudo /home/admin/config.scripts/blitz.datadrive.sh status)`
then https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/blitz.datadrive.sh#L402 calls
`echo "raidCandidate[${drivecounter}]='${mountoption}'"`
which gets it's output from the lines prior.
if i run `lsblk -o NAME,VENDOR | grep "^${disk}" | awk '$1=$1' | cut -d " " -f 2`
as in the script, it gives this output for me:
```
VENDOR
ASMT
`-sda1
SanDisk'
`-sdb1
mmcblk0
|-mmcblk0p1
`-mmcblk0p2
```
if i call`
sudo /home/admin/config.scripts/blitz.datadrive.sh status`
then the output is:
```
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=1
isSSD=0
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=1000203091968
hddGigaBytes=931
hddUsedInfo='387G (45%)'
hddDataFreeKB=506262456
hddAdapterUSB='174c:55aa'
hddAdapterUSAP=1

# RAID
isRaid=0
raidCandidate[0]='sdb 28 GB SanDisk''
raidCandidates=1

# SWAP
isSwapExternal=0
```
but if i call my 00test.sh containing:
```
#!/bin/bash
source <(sudo /home/admin/config.scripts/blitz.datadrive.sh status)
```
then the output is:
```
/dev/fd/63: line 21: unexpected EOF while looking for matching `''
/dev/fd/63: line 27: syntax error: unexpected end of file
```
There is some error happening while filtering the infos

rootzoll, 2021-09-07:
Thanks for the details ... my best guess is that the `disk` var is empty and this leads to parsing going south. I added a check on disk with `f [ ${devMounted} -eq 0 ] && [ "${disk}" != "${hdd}" ] && [ "${hdd}" != "" ] && [ "${disk}" != "" ]; then` ... but I am not able to test this.

This will be part of v1.7.1 release.

nicolasdiehl, 2021-09-10:
this did not fix it as my `vedorname` variable is `Sandisk'` somehow
the single quotation mark is causing the issue, if i add escaping with sed then it works:
https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/blitz.datadrive.sh#L400
`vedorname=$(lsblk -o NAME,VENDOR | grep "^${disk}" | awk '$1=$1' | cut -d " " -f 2 | sed 's/\x27/\x27"\x27"\x27/g')`

rootzoll, 2021-09-13:
Its a good idea to sanatize the vendorname ... instead escaping special characters I added a sed that will strip all non-alphanumeric characters from the string with `sed 's/[^a-zA-Z0-9]//g'` .. that should solve the problem more broadly.

rootzoll, 2021-10-10:
testing went well with the latest release candidates ... so closing this issue for final v1.7.1 release.

closed 2021-10-10 by rootzoll