raspiblitz/raspiblitz #2478: Mosh over Tor

nyxnor, 2021-08-13:
Mentioned by @xanoni here https://github.com/rootzoll/raspiblitz/pull/2419#issuecomment-890714473

Guide here https://matt.traudt.xyz/posts/mosh-over-tor-RczgjEqF/

xanoni, 2021-08-13:
Thank you. So my takeaway was that it's likely not to make sense, given Mosh opens a second connection to the server (besides the main SSH connection). This second connection is UDP-based and Tor does not support that.

However, it can be done over I2P if the C++ implementation (`i2pd`) is used, which natively supports UDP, no problem at all. I said I could work on a script for I2P, but it's somewhat difficult to test/debug until I set up an own Raspiblitz.

However, I can start with some simple instructions if that helps.

xanoni, 2021-08-13:
Oh, I forgot to address your guide ... I would personally not recommend that, because the UDP connection would go over clearnet ... thus there's really no benefit over standard SSH. Or am I missing something?

The other solution that he mentions was discussed in the Mosh repo, but didn't look positive: https://github.com/mobile-shell/mosh/issues/13#issuecomment-890730630

nyxnor, 2021-08-15:
regarding your points, closing.

closed 2021-08-15 by nyxnor