raspiblitz/raspiblitz #2479: Can’t cancel HTTPS subscription

Xill4X, 2021-08-14:
Hallo 🙋‍♂️,

When klick on cancel subscription for the HTTPS from duckdns.org it does some stuff, restarts Sphinx, but after it it’s still there.
Is there a way to delete it from the terminal out? 
thank you 🙏

rootzoll, 2021-08-16:
The script with which you can control the subscriptions is: `/home/admin/config.scripts/blitz.subscriptions.ip2tor.py`

If that is not working also you can just hard delete the file `/mnt/hdd/app-data/subscriptions/subscriptions.toml` .. so the background process will not renew any exiting IP2Tor subsciptions you have anymore.

cstenglein, 2023-06-07:
@Xill4X is this still relevant? Closing for now, if it's still relevant, please re-open

closed 2023-06-07 by cstenglein