raspiblitz/raspiblitz #2482: WebUI test for password A login

rootzoll, 2021-08-17:
If users login thru the WebUI they should also use password A - because thats feels natural, like using password A on SSH. At the moment we dont store password A in the clear anywhere - it is just set for the unix users and thats it.

So that we can use also password A for WebUI login we need for v1.8 a small script that can test if a given password A is correct - that is called by the API locally. Here is an idea how to make such a test: https://stackoverflow.com/questions/18035093/given-a-linux-username-and-a-password-how-can-i-test-if-it-is-a-valid-account

Also keep in mind that the API endpoint for login and testing the password should include a brute force protection.

nyxnor, 2021-09-14:
OnionShare stops serving a site after 20 failed passwords
something like that is good enough.

Also, password A is not using TLS scares me, but already discussed that the browser warnings about self signed certificate are difficult to deal with for new people.

rootzoll, 2021-09-21:
The plan is to push the user to use the TorBrowser to connect with the RaspiBlitz for setup & regular login - so those password transmission would be protected. There will be offered a fallback to login/setup with simple `http´ to local IP in normal browser ... but that is than a user choice to take the risk.

rootzoll, 2022-05-10:
The password A login is now possible with the BlitzAPI. To cut complexity and get the WebUI out of the door we will start with also with simple unsecured http webui connection for v1.8.0 - but offering secured Tor is on the list for v1.8.1: see #3105

rootzoll, 2022-06-05:
Closing this genreal one for v1.8.0 release with this #3145 error issue still open.

closed 2022-06-05 by rootzoll