raspiblitz/raspiblitz #2484: Copy blockchain in the computer by connecting the target and source disks

tifDev, 2021-08-18:
After trying to copy over lan I noticed that it basically does an `rsync`.

I want to know if it's possible to connect the disks on my computer and just do the `rsync` on my computer instead of over lan.

Example:
I connect both the source and target disk of the RPI's in my PC, and then I do `rsync` from source disk (containing the blockchain) to the target disk.

Is this possible? If so, what are the directories and the commands that should be run? (I presume that some extended attributes should be copied along with the files)

Thank you.

rootzoll, 2021-08-19:
Basically you could do that ... but we dont advise this as regular praxis because most people have Windows or Mac laptops that cannot read/write EXT4 partitions - thats why copying over LAN is the most compatible option to sync the Blockchain data.

closed 2021-08-19 by rootzoll

tifDev, 2021-08-19:
In this case it would be in a PC with Linux. Could you advise the rsync command?

These are the files that my BLOCKCHAIN HDD has:
``` bash
lost+found/
litecoin/
ssh/
app-storage/
app-data/
swapfile
lnd/
temp/
raspiblitz.conf*
bitcoin/
```

1. Are all the files needed to be copied in the rsync command? If not which files are needed?
2. Any files that need also to be copied or require any modification in the SD card?

Thank you