raspiblitz/raspiblitz #2486: log files getting bigger and bigger

KoreaComK, 2021-08-22:
Hi,

My log files are increasing a lot. I see with some friends of mine that they have a log `/var/log/` with less than 50MB, however, this is what I receive when I use the commando to see the size of folder:

```
admin@raspberrypi:/var/log $ sudo du -h /var/log/
60K     /var/log/apt
36K     /var/log/cups
4.0K    /var/log/private
72K     /var/log/nginx
786M    /var/log/
```

When I look inside of the folder that is given as a return:

```
total 804740
drwxr-xr-x  6 root root      4096 Aug 19 00:00 .
drwxr-xr-x 12 root root      4096 Jun 24 08:43 ..
drwxr-xr-x  2 root root      4096 Aug  9 20:00 apt
-rw-r-----  1 root adm   16332948 Aug 21 23:59 auth.log
-rw-r-----  1 root adm  320325508 Aug 21 16:45 auth.log.1
-rw-r-----  1 root adm    4522135 Aug 14 22:04 auth.log.2.gz
-rw-rw----  1 root utmp         0 Aug  2 18:34 btmp
drwxr-x---  2 root lp        4096 Aug 21 00:00 cups
-rw-r-----  1 root adm    9484489 Aug 21 23:59 daemon.log
-rw-r-----  1 root adm  238753434 Aug 21 16:45 daemon.log.1
-rw-r-----  1 root adm    8333494 Aug 14 22:04 daemon.log.2.gz
-rw-r-----  1 root adm      18106 Aug 21 16:47 debug
-rw-r-----  1 root adm       1646 Aug 10 20:51 debug.1
-rw-r--r--  1 root root         0 Aug  9 20:00 dpkg.log
-rw-r--r--  1 root root      1028 Aug  8 23:20 dpkg.log.1
-rw-r-----  1 root adm      22083 Aug 21 16:47 fail2ban.log
-rw-r-----  1 root adm       2099 Aug 10 20:51 fail2ban.log.1
-rw-r-----  1 root adm   17343442 Aug 21 23:59 kern.log
-rw-rw-r--  1 root utmp    298368 Aug 21 17:14 lastlog
-rw-r-----  1 root adm    6560572 Aug 21 23:59 messages
-rw-r-----  1 root adm    3583193 Aug 14 22:04 messages.1
drwxr-xr-x  2 root root      4096 Aug 20 00:00 nginx
drwx------  2 root root      4096 Aug  2 18:34 private
-rw-r-----  1 root adm    9784538 Aug 21 23:59 syslog
-rw-r-----  1 root adm  177952368 Aug 21 16:45 syslog.1
-rw-r-----  1 root adm    1022690 Aug 16 20:11 syslog.2.gz
-rw-r-----  1 root adm    6197592 Aug 21 23:59 ufw.log
-rw-r-----  1 root adm    3548985 Aug 14 22:04 ufw.log.1
-rw-r-----  1 root adm      11561 Aug 21 16:47 user.log
-rw-r-----  1 root adm        232 Aug 10 20:51 user.log.1
-rw-rw-r--  1 root utmp    116000 Aug 21 17:14 wtmp
```

When I open the `auth.log`, it keep spamming the same lines:
```
admin@raspberrypi:/var/log $ sudo tail -f auth.log
Aug 22 00:00:11 raspberrypi sudo: pam_unix(sudo:session): session opened for user bitcoin by pi(uid=0)
Aug 22 00:00:11 raspberrypi sudo: pam_unix(sudo:session): session closed for user bitcoin
Aug 22 00:00:11 raspberrypi sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=bitcoin ; COMMAND=/usr/local/bin/lncli --macaroonpath=/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/readonly.macaroon --tlscertpath=/home/bitcoin/.lnd/tls.cert feereport
Aug 22 00:00:11 raspberrypi sudo: pam_unix(sudo:session): session opened for user bitcoin by pi(uid=0)
Aug 22 00:00:11 raspberrypi sudo: pam_unix(sudo:session): session closed for user bitcoin
Aug 22 00:00:17 raspberrypi sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/ls /var/cache/raspiblitz/info.json
Aug 22 00:00:17 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:17 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Aug 22 00:00:21 raspberrypi sudo:    admin : TTY=pts/0 ; PWD=/var/log ; USER=root ; COMMAND=/usr/bin/tail -f auth.log
Aug 22 00:00:21 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by admin(uid=0)
Aug 22 00:00:22 raspberrypi sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/home/admin/config.scripts/internet.sh status
Aug 22 00:00:22 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:22 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Aug 22 00:00:22 raspberrypi sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/ls /home/admin/recover.flag
Aug 22 00:00:22 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:22 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Aug 22 00:00:22 raspberrypi sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=bitcoin ; COMMAND=/usr/local/bin/lncli --chain=bitcoin --network=mainnet getinfo
Aug 22 00:00:22 raspberrypi sudo: pam_unix(sudo:session): session opened for user bitcoin by pi(uid=0)
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session closed for user bitcoin
Aug 22 00:00:23 raspberrypi sudo:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/home/admin/config.scripts/blitz.datadrive.sh status
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:23 raspberrypi sudo:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/df
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Aug 22 00:00:23 raspberrypi sudo:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/btrfs filesystem show
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
Aug 22 00:00:23 raspberrypi sudo:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/ls -l /mnt/hdd
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Aug 22 00:00:23 raspberrypi sudo: pam_unix(sudo:session): session closed for user root
```

The `daemon.log` gives me that:
```
admin@raspberrypi:/var/log $ sudo tail -f daemon.log
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 8 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 9 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 10 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 11 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 12 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 13 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 14 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Fetched transaction 15 / 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: New mempool size: 1210 Change: 15
Aug 22 00:01:20 raspberrypi node[1695]: Aug 22 00:01:20 [1695] DEBUG: Mempool updated in 0.164 seconds
Aug 22 00:01:22 raspberrypi node[1695]: Aug 22 00:01:22 [1695] DEBUG: Updating mempool
Aug 22 00:01:22 raspberrypi node[1695]: Aug 22 00:01:22 [1695] DEBUG: Fetched transaction 1 / 2
Aug 22 00:01:22 raspberrypi node[1695]: Aug 22 00:01:22 [1695] DEBUG: Fetched transaction 2 / 2
Aug 22 00:01:22 raspberrypi node[1695]: Aug 22 00:01:22 [1695] DEBUG: New mempool size: 1212 Change: 2
Aug 22 00:01:22 raspberrypi node[1695]: Aug 22 00:01:22 [1695] DEBUG: Mempool updated in 0.031 seconds
```

I tried to see find a solution but I couldn't find anyone. Does you guys know what is going on?

xanoni, 2021-08-23:
You can manually delete the older files (wouldn't delete the most recent versions, as that can sometimes cause permission problems when daemons can't re-create them themselves).

The cleaner way would be to tweak the settings in `/etc/logrotate.conf`, as well as the files in `/etc/logrotate.d/` and then do a `sudo systemctl restart logrotate`.

You'll see lines like `weekly` and `rotate X` that determine how often logs are rotated and how may logs are kept.

rootzoll, 2021-08-23:
@frennkie did we lost the changes that keep the log files small (on rotate) automatically?

rootzoll, 2021-09-23:
OK with v1.7.1rc2 there on setup/recovery the `/etc/logrotate.conf`will be changed to rotate daily, keep 2 weeks of old logs & compress them.